arendst/Tasmota #241: Tasmota 4.0.7 won't work on NodeMCU/WeMos

crazynet24, 2017-03-20:
I've tried to upgrade from 4.05 to 4.07 at a NodeMCU (ESP12E, 4MB), but it hangs after uploading the new image. After reseting it hangs in a loop while showing the CRC values in the serial output, no more messages. When reverting to 4.05, it works normal.

mihalski, 2017-03-20:
I installed it on both the (new) standard Wemos and the pro version without any issues.

crazynet24, 2017-03-20:
Thanks mihalski!
I tried it with 4.05, 4.06 and 4.07 again, same compiler settings in arduino IDE (NodeMCU 1.0, 4M/3SPIFFS), always the same user-settings file (added Port 80 to the newer one), but I always get a boot loop with this output:
--------------------------------
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
-----------------------------------

Only version 4.05 is working for me, versions above are not more working anymore. My logic ends here...

mihalski, 2017-03-20:
Have you tried downloading the precompiled firmware?

https://github.com/arendst/Sonoff-Tasmota/releases/download/v4.0.7/firmware.bin

dc75, 2017-03-20:
I have version 4.07 working on a NodeMCU board with no problems.  Uploaded the firmware using Arduino IDE with the same settings as listed in the wiki.

I had a problem similar to this some time ago(sorry, don't remember the exact error), but it was due to an IP address conflict on my network; once I corrected that, everything worked fine.

crazynet24, 2017-03-20:
Thanks for answering!
@mihalski : no, I've newer known about the precompiled versions. 

@dc75 my module doesn't connect anymore to wlan with versions >4.05. 

I will see... perhaps it will be clear somedays, at the moment I can work with 4.05

closed 2017-03-27 by rhodopsik