arendst/Tasmota #242: empty value handling

evilgreen, 2017-03-20:
I have a few parameters in user_config.h which are left empty, for example:
`#define MQTT_USER            ""       // [MqttUser] Optional user`
`#define MQTT_PASS            ""       // [MqttPassword] Optional password`
`#define NTP_SERVER2            ""    // [NtpServer2] Select second NTP server by name or IP address (5.39.184.5)`
`#define NTP_SERVER3            ""  // [NtpServer3] Select third NTP server by name or IP address (93.94.224.67)`

This empty settings will be reflected in the actual device configuration, so no problem yet.
If I will ever upgrade using the pre-complied image from github zip I will see such parameters overridden by defaults like "DVES_USER" for MQTT username, etc.
Is that an intended behavior? From my perspective my empty values should be preserved as all others.

And a side question - is it possible to clear (or set empty) values from console or via MQTT? If yes - what's the syntax then? Using "" will produce """".

arendst, 2017-03-20:
Have you tried to upgrade using a pre-compiled image and did you notice the behaviour you expect?

evilgreen, 2017-03-20:
@arendst This is what I'm saying - after upgrading from user-compiled to pre-compiled image from github my empty defaults got overridden by your defaults. As I recall from my last night testing, MQTT_USER and MQTT_PASS were overridden while NTP_SERVER2 and NTP_SERVER3 were not.
I can re-test again if necessary, just give me instructions. Thanks!

arendst, 2017-03-20:
No need to test. I'll check

arendst, 2017-03-21:
Two items to discuss.

Resetting your previous empty values after loading the pre-compiled image is expected as long as your CFG_HOLDER define is different from the pre-compiled define. In that case all settings are reset to the pre-compiled values. If you want them to stay as you like you will have to reset your CFG_HOLDER to the same value as the pre-compiled value. In that case your empty fields stay empty as no settings are reset.

About setting empty values via MQTT the only possible solution I see is sending a 0 for an empty field which I will translate to "".

evilgreen, 2017-03-21:
Using zero for an empty value looks good for me.
Here is my test plan:

1. compiled a binary with "#define CFG_HOLDER 0x20161209" and some empty values
2. loaded that binary from the web gui
3. "reset 1" from the console to load my defaults
4. checking for empty values - all good
```
CMND: mqttuser
MQTT: stat/sonoff/RESULT = ["MqttUser":""}
```
5. OTA with the github pre-compiled image
6. checking for previously empty values - all good
7. changing topic from _sonoff_ to _sonoff2_
8. checking for previously empty values - MqttUser is not empty anymore:
```
CMND: mqttuser
MQTT: stat/sonoff2/RESULT = ["MqttUser":"DVES_USER"}
```

BTW, what's the meaning of "DVES"?

arendst, 2017-03-21:
Do not forget to mention you do this all via the web interface as using the commands via console or MQTT will not do what you experience. Anyway. The latest release (4.0.8) should solve your problem.

Having a MQTTuser and/or password should not be any problem if you have configured your MQTT broker for not using MQTTuser and/or password so that's why no one raised the issue before.

The default names DVES_ stands for DeVice ESp8266 the processor chip inside sonoff and was used by my initial MQTT client designed by Tuan (https://github.com/tuanpmt/esp_mqtt).

evilgreen, 2017-03-21:
MQTTuser value was taken as an example only as it was easily noticeable in the web gui. It's clear that your default username will not harm and now it could be even cleared by using zero value. Tested 4.0.8 - all looks good, thanks a lot!

closed 2017-03-21 by evilgreen