arendst/Tasmota #243: Intermittent problem with OTA upgrade (URL becomes truncated)?

ecsfang, 2017-03-20:
Hi,

I have specified an OTA URL to a local machine as default config in my build, and sometimes when doing an OTA upgrade, the upgrade is unsuccessful. That would be ok (bad WiFi or waht ever), but sometimes something happens and after the attempt the OTA URL is corrupt. It seems as if only the first 8 characters of the address remains (in all cases I have seen, it always looks the same):

    OTA Url:
       http://1

In the last case, I saw the following reason for the failed upgrade:

    stat/garden/UPGRADE : msg.payload : string [37]
        **Failed HTTP error: connection refused**

The upgrade was done via the web-interface, and the URL looked ok before I started,
I have seen this behavior on both Sonoff Led and 4CH.
Updating the URL again solves the problem, and the next try is successful.

arendst, 2017-03-20:
#235 explains the situation which should have been solved with the latest release 4.0.7

ecsfang, 2017-03-20:
Sorry - missed that one!
And I was updating from 4.0.5 in my case so the bug was still present! :)

Cheers!

closed 2017-03-20 by ecsfang