arendst/Tasmota #245: Switchtopic and MQTT behaviour

mihalski, 2017-03-21:
Here's the setup:
- Wemos
- Pushbutton (as switch1)
- PIR (as switch2)

I've discovered that switchtopic cannot be set to the same topic as the main MQTT topic or this causes COMMANDS to flood the logs. I assume this is some sort of loopback. So say we have the mqtt topic master_bedroom and switchtopic is set to master_bedroom_switch.

Upon detecting movement PIR goes high and we get:
```
18:54:12 MQTT: cmnd/master_bedroom_switch/POWER2 = ON
18:54:14 MQTT: cmnd/master_bedroom_switch/POWER2 = OFF
```
Neither of these state changes appears at stat/master_bedroom_switch/POWER2 or stat/master_bedroom/POWER2.

Should this work as I expect or am I misunderstanding how this is supposed to function.

davidelang, 2017-03-21:
It is working as designed.

tele/ are reports of current status

stat/ are reports that the device just changed it's state

cmnd/ commands

when you don't have a switchtopc/buttontopic, the sonoff switches the relay and 
then reports the change (stat/)

if you send a cmnd from some other source, the sonoff switches the relay and 
reports the change (stat/)

periodically, it sends telemetry (tele/) showing what it's current state

if you set a switchtopic/buttontopic, you are telling it that you don't want to 
change the relays directly, instead you want to just send a mqtt message. It 
can't be a stat/ or tele/ message because it didn't change the relays, so it is 
a cmnd message instead

On Tue, 21 Mar 2017, mihalski wrote:

> Here's the setup:
> - Wemos
> - Pushbutton (as switch1)
> - PIR (as switch2)
>
> I've discovered that switchtopic cannot be set to the same topic as the main MQTT topic or this causes COMMANDS to flood the [...]

mihalski, 2017-03-21:
Thank you.

While not intuitive it does follow some sort of logic.

What would you say is the "right" way of handling my situation? Simply set the state topic of the clients to point at cmnd/master_bedroom_switch/POWER2 ?

davidelang, 2017-03-21:
I'm not exactly sure what you are trying to do.

The intended use of buttontopic/switchtopic is that they will send messages to 
your mqtt server, and your home automation software will look for these 
messages, apply some sort of rules to them, and then issue commands as 
appropriate (switching on other lights for example)

mihalski, 2017-03-21:
A PIR is a Passive infrared sensor (https://en.wikipedia.org/wiki/Passive_infrared_sensor). I am using it as a motion detector. The push button is a simple momentary switch. Both of these are talking to Home Assistant (http://homeassistant.io), an open-source home automation platform.

The logic in Home Assistant is besides the point of the question, I just need to know which MQTT topic I need to monitor to allow me to trigger my home automation. Am I supposed to be subscribing to the (in the case of no relays) pseudo "state topic" cmnd/master_bedroom_switch/POWER2  to detect motion? (And thus cmnd/master_bedroom_switch/POWER1 to detect the momentary switch).

davidelang, 2017-03-21:
> The logic in Home Assistant is besides the point of the question, I just need
> to know which MQTT topic I need to monitor to allow me to trigger my home
> automation. Am I supposed to be subscribing [...]

Yes.

mihalski, 2017-03-21:
Thank you.
This might help clear up some other problems I was having.

closed 2017-03-21 by mihalski