arendst/Tasmota #246: Sonoff Touch sensor

joshuagti, 2017-03-21:
I'm not sure if it is a firmware thing or a device, but since pulling the unit from the wall and reflashing it with Tasmota, the touch sensor will not trigger unless i pull the faceplate off and touch the sensor directly. Is there any way to modify sensitivity as such? 

Thanks

arendst, 2017-03-21:
Nope, it's pure mechanical. Your pulling did it.

ionciubotaru, 2017-03-21:
The touch circuit has a dedicated IC connected to esp's GPIO0.
It works without plastic shield.
The touch it is not so fast to use it to put esp in programming mode - you have to put it manually for the first time programing

davidelang, 2017-03-27:
closing this issue as it does not seem to be a software issue. please reopen this issue if you think there is something that can be done in software for this.

closed 2017-03-27 by davidelang

zaum, 2025-03-25:
In case someone runs into this problem... Attach a conductive material (copper foil or aluminum tape) between the touch sensor pad and the plastic cover. Its a hardware issue.