arendst/Tasmota #247: Is it possible to add input (button, switch) that is not connected to any relay?

KonstantinMastak, 2017-03-21:
Maybe I'm missing something, but as far as I see in code, it is only possible to introduce external button or switch only if it is hard-connected with on-board relay with some function (i.e, follow, invert, toggle, etc.). But what if I want to simply have a digital input which publishes its state to MQTT but is not connected to any relay onboard? For example, to control external relay, to let broker decide what to do with it.

Is there any way to do this without code changes? If not, where is the best place in code to put change to make code impact as low as possible?

Thank you!

arendst, 2017-03-21:
Did you try to add a switch, set switchtopic to a MQTT topic you like and see what happens?

khcnz, 2017-03-21:
Read up here https://github.com/arendst/Sonoff-Tasmota/issues/245 and here https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

KonstantinMastak, 2017-03-21:
Thank you very much! This answers my question.

closed 2017-03-21 by arendst

aihysp, 2018-07-02:
Hi Theo can you please explain me in stupid terms, how can i disconnect my GPIO 14 sensor from the relay
i am trying to setup a sensor , that is not related to the relay but in tasmota i am getting same topic changed when i move near the sensor

i use sonoff basic with some 0516 rader sensor 

please help me configre this :)

daywalkerSi, 2018-09-19:
I made a simple input for myself. I detailed it here - isonoff.wordpress.com

packetbrain, 2018-09-25:
For grins, I did this on the workbench... an alternate way to do this - gets initial state within teleperiod and real-time changes too... no relay pins required

`-------------------------------------------------------------------------------------------------------------
rule1 on switch2#state=1 do publish tele/wemos1/SENSOR {"Switch2":"ON"} endon on switch2#state=0 do publish tele/wemos1/SENSOR {"Switch2":"OFF"} endon on switch3#state=1 do publish tele/wemos1/SENSOR {"Switch3":"ON"} endon on switch3#state=0 do publish tele/wemos1/SENSOR {"Switch3":"OFF"} endon on switch4#state=1 do publish tele/wemos1/SENSOR {"Switch4":"ON"} endon on switch4#state=0 do publish tele/wemos1/SENSOR {"Switch4":"OFF"} endon  
-------------------------------------------------------------------------------------------------------------`

and in HomeAssistant:
-------------------------
`binary_sensor:
  - platform: mqtt
    name: "Wemos Sw 2"
    state_topic: "tele/wemos1/SENSOR"
    value_template: '{{ value_json["Switch2"] }}'

  - platform: mqtt
    name: "Wemos Sw 3"
    state_topic: "tele/wemos1/SENSOR"
    value_template: '{{ value_json["Switch3"] }}'

  - platform: mqtt
    name: "Wemos Sw 1"
    state_topic: "stat/wemos1/POWER"`

NOTE: Wemos Sw 1 has an attached relay... so using the regular state method... no rules or value_template required