arendst/Tasmota #248: Temperature and Humidity Sensors

raisude, 2017-03-21:
Just one question: when attaching a  DHT22 Sensor to the Sonoff Basic should there be an Output of the values on the startpage of the Webserver? https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration#i2c-sensor Shows something like that. But I have configured one DHT22 but I don`t have such Output.

Regards Rainer

davidelang, 2017-03-21:
Yes, once you have set the gpio setting to say that a specific gpio is connected 
to a sensor, you should start seeing that sensor data in the logs (and on the 
main webserver page). note that the logs are set by default to be reported every 
5 min.

You may have to restart the device after you have configured it.

On Tue, 21 Mar 2017, raisude wrote:

> Just one question: when attaching a  DHT22 Sensor to the Sonoff Basic should there be an Output of the values on the startpage of the Webserver? [...]

decologne, 2017-03-21:
To get more information of the working of DHT, connect your device by webserver, set Logging webserver in mode More debug, start Webserver Console and restart your device. Now you get more detail info.

lkhaskjdhlkashd, 2017-03-22:
I have a similar problem with a "dallas 18B20" temp sensor, i enabled the option USE_DS18x20 in user_config.h

Connected the sensor directly to the Sonoff Basic on pin 1-4-5
1 - VCC
4 - GND
5 - DATA

Is there something i forgot to do?

I noticed that there is an internal temperature sensor... how can i enable that?

raisude, 2017-03-22:
Are you sure you`re using the right pins. In my case the problem was that I inverted VCC and ground. This because of there are some wrong pictures in the web. And pin 1 VCC is the only one not round but square coded on the board.

lkhaskjdhlkashd, 2017-03-22:
Pretty sure --> [https://i.imgur.com/RRA6LOw.jpg](url)

raisude, 2017-03-22:
and in configure menu you`re using gpio14 Type4 18B20. And the other gpios are none or at least no sensor type of the same group as the 18B20?

lkhaskjdhlkashd, 2017-03-22:
I cant select Type4 18B20,

See the link for an image --> [https://i.imgur.com/wItmIpL.png](url)

davidelang, 2017-03-22:
what version are you running?

on current versions you need to configure the gpio that you have the sensor 
connected to to the appropriate sensor type

I'm not sure what internal temp sensor you are talking about.

lkhaskjdhlkashd, 2017-03-22:
It works, i forgot the resistor,,, :-|

davidelang, 2017-03-27:
closing this issue as it's now working

closed 2017-03-27 by davidelang