arendst/Tasmota #249: HomeKit emulation

ageorgios, 2017-03-21:
Hello,
is there the ability to implement HomeKit emulation as a feature of Tasmota?

Here is the way to have native HomeKit on ESP8266
https://github.com/HomeACcessoryKid/ESP8266-HomeKit

Thanks for the great work.

davidelang, 2017-03-21:
I would not consider it likely, the esp8266-homekit code is about as large as 
the entire current codebase.

I took a quick look and found no documentation for the protocol itself, just 
pointers to the libraries that implement it.

On Tue, 21 Mar 2017, ageorgios wrote:

> Hello,
> is there the ability to implement HomeKit emulation as a feature of Tasmota?
>
> Here is the way to have native HomeKit on ESP8266
> https://github.com/HomeACcessoryKid/ESP8266-HomeKit
>
> Thanks for the great work.
>
>

khcnz, 2017-03-21:
It is reasonably straight forwards to connect tasmota to OpenHab using MQTT and then use the Openhab Homekit binding. This way I can control my sonoffs with Homekit integration.

ageorgios, 2017-03-21:
The way I meant is controlling sonoff through homekit without any kind of bridge

MacWyznawca, 2017-03-21:
Better idea is use MQTT Broker and HomeBridge.
I made sam plugins for Tasmota fimrware:
https://www.npmjs.com/~macwyznawca

davidelang, 2017-03-21:
given that the homekit for esp8266 can't load all it's features at once because 
it runs out of memory, and tasmota can't enable TLS and the webserver at the 
same time because of lack of memory, I don't see any way that it would be 
possible to combine the two and have the result use less memory than either.

ageorgios, 2017-03-21:
@MacWyznawca 
great work!
I have implemented my own plugin for homebridge using the http commands:
https://github.com/ageorgios/homebridge-sonoff-tasmota-http
what is the added value for an mqtt server + homebridge?

@davidelang 
You may be right I do not know the memory footprint of the things you say, maybe if it disabled mqtt protocol and let only http on homekit emulation?

davidelang, 2017-03-21:
If you disable mqtt from tasmota, what is the advantage of adding homekit 
emulation to it rather than just using the esp-8266-homekit firmware?

The page listed earlier says that the esp8266-homekit firmeware can't load many 
modules until after it finishes connecting, due to it's memory restrictions.

Right now, disabling mqtt in tasmota doesn't actually comment out any code, it 
just doesn't run much of it.

MacWyznawca, 2017-03-21:
A lot ...  😉
you can check if the device is active (LWT command with MQTT broker), more responsiveness. Generlanie more possibilities of communication in both directions.

davidelang, 2017-03-21:
@MacWyznawca It's not clear what you are responding to.

closed 2017-03-27 by davidelang