arendst/Tasmota #251: SONOFF POW - Add reactive, active, apparent power etc. - like in espurna software

PitPaul, 2017-03-21:


arendst, 2017-03-22:
Here are the results using XosePerez calculations:

- His Active Power is the result shown as Power
- His Apparent Power is the multiplication of Voltage and Current
- His Reactive Power is either 0 if Power Factor = 1 or the Square root of Square Apparent - Square Active

So the results are there, you just have to calculate it yourself :-)

PitPaul, 2017-03-22:
Ok, so it is simple electrotechnic formula - thanx

closed 2017-03-22 by PitPaul

khcnz, 2017-03-22:
Hi @PitPaul perhaps you could detail on the wiki for others in the future? https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow

andrethomas, 2018-09-28:
@khcnz I think for people who are confused there is always this https://en.wikipedia.org/wiki/AC_power