arendst/Tasmota #253: Upgrade from 4.0.7 not working

gthb96, 2017-03-21:
Hello,
I am trying to upgrade my Sonoff from 4.0.7 via http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin but it just reboots with the old firmware again.
The restart reason says "RSLT: INFO3 = {"Started":"Blocked Loop"}"

How can I get to the latest firmware?
If I need to manually upload it on the webinterface do I need to use "sonoff.ino.bin or the firmware.bin?
What is the difference between those two as they are different file sizes

PitPaul, 2017-03-21:
Best way to get latest firmware is to compile source with Platformio
After compilation,  file is usually in folder /Platformio/Sonoff-Tasmota4.0.7/.pioenvs/sonoff

gthb96, 2017-03-21:
But does it make a difference if I recompile it everytime or if I just use firmware.bin or sonoff.ino.bin?

PitPaul, 2017-03-21:
Settings (those from user_config.h) are stored in flash, and should be preserved.When not - go to settings mode with button.  (https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage)

gthb96, 2017-03-21:
But when I flash a precompiled version and reset the device to factory defaults I lose my Wifi for example or not?

PitPaul, 2017-03-21:
1 flashing do **not change** the your settings 
2. factory reset, **turn device in default settings** - precompiled from stored parameters in user_config.h file.

gthb96, 2017-03-21:
And do you know the difference between the firmware.bin and sonoff.ino.bin which are precompiled?

PitPaul, 2017-03-21:
Its look like booth of them are precompiled - firmware.bin is file from Platformio, and sonoff.ino.bin is  from arduino

gthb96, 2017-03-21:
Thank you very much for your explanation!

closed 2017-03-21 by gthb96