arendst/Tasmota #254: Error: Please specify `upload_port` for environment or use global

HansWurst90, 2017-03-21:
HI,

I followed the whole tutorial and did everything as explained. That's the error I get when trying to install 4.0.8 with platform.io

```
Checking program size
Building .pioenvs/sonoff/firmware.bin
error: can't read ELF file header
error: wrong header for ELF file!

*** [.pioenvs/sonoff/firmware.bin] Error 255
========================== [ERROR] Took 60.80 seconds ==========================
Checking program size
Building .pioenvs/sonoff/firmware.bin
error: can't read ELF file header
error: wrong header for ELF file!
*** [.pioenvs/sonoff/firmware.bin] Error 255
========================== [ERROR] Took 61.57 seconds ==========================
Checking program size
text       data     bss     dec     hex filename
460257    10060   34000  504317   7b1fd .pioenvs/sonoff/firmware.elf
Looking for upload port...
Building .pioenvs/sonoff/firmware.bin

Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
========================== [ERROR] Took 61.81 seconds ==========================
========================== [ERROR] Took 61.74 seconds ==========================
```

Also, another Question. I have to admit I'm fairly new to MQTT. I am using homebridge and want to use the sonoff together with the homebridge [plugin](https://github.com/suculent/homebridge-plugin-sonoff)

in the [requirements](https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite) it says I need a MQTT-Broker and Client. Does the Homebridge act as the Broker or wow exactly does that work?

Thank's in advance.

Jan

MacWyznawca, 2017-03-22:
Try to close, and open Atom.

You check [„Serial monitor”](https://mojmac.pl/wp-content/uploads/2017/03/Sprawdzenie-adaptera-USB-TTL.png)

Perhaps, your usb-rs adapter is not seen by the system. What model do you have?

HomeBridge don't have MQTT broker, You must instal it extra.

See these articles (in Polish, use translator):
https://mojmac.pl/2017/03/11/inteligentny-dom-tanim-kosztem-homekit-zrob-to-sam-cz-3/

https://mojmac.pl/2017/03/18/inteligentny-dom-tanim-kosztem-homekit-mqtt-zrob-to-sam/

HansWurst90, 2017-03-22:
[This](https://www.amazon.de/gp/product/B00QFDRPVI/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1) is the USB-Adapter I'm using.
I will check the Serial Monitor when I'm home and will look into how to install the Broker.
Thanks for your tips so far

MacWyznawca, 2017-03-22:
MacOS? Witch this model are some problems. Better use with FT232

HansWurst90, 2017-03-22:
Yes, MacOS.
I did not now, there were differences between adapters.
I will give it a try tonight. I it's not working, I will try the adapter you recommended.
Would [this one](https://www.amazon.de/Adapter-FT232RL-Arduino-Christians-Technikshop/dp/B0178HVEH0/ref=sr_1_1?ie=UTF8&qid=1490170473&sr=8-1&keywords=FT232) be compatible?

MacWyznawca, 2017-03-22:
I use this:
http://elty.pl/pl/p/Adapter-USB-na-UART-z-FT232-zlacze-micro-USB/1716

davidelang, 2017-03-27:
did you get this working?

HansWurst90, 2017-03-27:
@MacWyznawca yes.
I had to find the right drivers for my serial adapter

closed 2017-03-27 by davidelang