arendst/Tasmota #255: MQTT topics are hardcoded

sboldt1978, 2017-03-21:
Hello,

I really like the Sonoff-Tasmota projekt and do appreciate your work very much. 
But there is one issue I came across: 
I use SUB_PREFIX, PUB_PREFIX and PUB_PREFIX2 values that differ from the default ones. But because they are not configurable I have to change them manually with every new release and compile the firmware myself before uploading it to my devices. 
I would really like the MQTT topics to be configurable through the web interface and to survive firmware firmware updates. This would be nice to have not only for the three topic-substrings listed above but for any MQTT topics used by the firmware.

Thanks a lot in advance!

davidelang, 2017-03-21:
Just a note that we are running out of ram on the current firmware, changing things from compiled-in constants that can sit in flash to runtime configurable things that have to sit in ram is a bit hard to justify.

to avoid having to manually edit user_config.h for every build, look at user_config_override.h. It's still part of the git tree, so you have to copy your saved version over what you download each time, but it's not going to change in the git repo.

arendst, 2017-03-22:
I'll make the change but only allow these items to be up to 7 characters each.

sboldt1978, 2017-03-22:
Thanks a lot!

arendst, 2017-03-25:
Implemented in version 4.1.0 just released.

closed 2017-03-25 by arendst

sboldt1978, 2017-03-25:
Where in the webinterface can you configure them? I couldn't find it

arendst, 2017-03-25:
See release notes and wiki. Use console.

Hogesyx, 2017-03-25:
@arendst I dont understand why you implement this to solve a non-issue(already can be done using override) and waste memory, and refuse to implement a flexible topic name #define that can be override.