arendst/Tasmota #256: Integration of ArduinoOTA Library

sboldt1978, 2017-03-21:
Hi,

it would be very handy to be able to update the devices directly OTA from E.G. the PlatformIO / Atom application. This is possible through the integration of the ArduinoOTA library which is really simple: 

In setup function: 
    ArduinoOTA.setPort(8266);
    ArduinoOTA.begin();

and somewhere in the loop:

  ArduinoOTA.handle();

That's all! I implemented it myself in my copy of Tasmota but  have to insert it time and time again after updating to the latest version. It would be nice to have this implemented in the Tasmota project itself (may be toggleable through the webinterface).

Thanks a lot in advance!

davidelang, 2017-03-21:
how is this different from the existing OTA support?

how large is this library? we are running into space problems already.

sboldt1978, 2017-03-21:
AFAIK the current OTA support only implements a way to either grab a firmware image via http or upload it through the web interface. With the ArduinoOTA library you can directly upload the firmware to the ESP as part of the build process.. I forgot to add that this is also possible with the Arduino GUI itself. The ESPs wich are online and have this library implemented are listed as "Ports" in the Arduino IDE. InATOM IDE you have to add a line to the platform IDE which points to the ip-address of the device you want to update: 

upload_port = 192.168.222.132

Another example can be found here: https://www.fais-le-toi-meme.fr/en/electronic/tutorial/esp8266-arduinoota-sketch-upload-over-the-air

The compiled ArduinoOTA library is roughly 44k in size.

davidelang, 2017-03-21:
given that every time you change the port in the arduino IDE it forces a (slow) 
recompile of everything before doing the upload, I would sure hate to use that 
to update more than one device.

IoT things have a reputation of being horribly insecure, the idea of listening 
on a network port and blindly installing whatever is passed to you seems like a 
great way to earn that reputation.

A script to upload a file to a web interface is fairly trivial to write (and is 
insecure enough, only an optional password to protect it), why is your approach 
better?

On Tue, 21 Mar 2017, sboldt1978 wrote:

> Date: Tue, 21 Mar 2017 15:47:03 -0700
> From: sboldt1978 <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f7dbe42adb240957f68d32cf92bf155e57908657992cf0000000114e96f6
>     792a169ce0cde59da@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] Integration of ArduinoOTA Library (#256)
> 
> AFAIK the current OTA support only implements a way to either grab a firmware image via http or upload it through the web interface. With the ArduinoOTA library you can directly upload the firmware to the ESP as part of the build process.. I forgot to add that this is also possible with the Arduino GUI itself. The ESPs wich are online and have this library implemented are listed as "Ports" in the Arduino IDE. InATOM IDE you have to add a line to the platform IDE which points to the ip-address of the device you want to update:
>
> upload_port = 192.168.222.132
>
> Another example can be found here: https://www.fais-le-toi-meme.fr/en/electronic/tutorial/esp8266-arduinoota-sketch-upload-over-the-air
>
> The compiled ArduinoOTA library is roughly 44k in size.
>
>

ecsfang, 2017-03-22:
Also,  I can upgrade ALL my 10+ devices with just one MQTT-command! Or a single device if that is needed. Much easier than having to update the port in the IDE.

arendst, 2017-03-22:
I was never able to use ArduinoOTA as I never saw to ports pop up in the IDE. Just abandoned it a long time ago...

sboldt1978, 2017-03-22:
I'm working with ArduinoOTA on all (10+) my ESP8266 devices and it works absolutely flawless. May be it improved vastly since you last tried to use it and give it another try. Regarding security of IOT devices: The option to update a firmware with just by sending something to an open port IS an issue, but IMO this is only true for devices that can be publicly accessed. I'm just using it to update my devices in my private Wifi network and I do not see any security issues here.

Of course this is not a good way to deploy a new firmware to many devices! But it comes in very handy when developing. You can deploy the changes you just made directly to one device and test it very easily. That's the main purpose of this OTA update.

davidelang, 2017-03-22:
what I do is publish to a webserver on my laptop (setting the destination in the 
arduino IDE to OTA)

then updating a device I'm testing is a simple compile (auto-published to the 
website) followed by triggering an OTA upgrade in the tasmota firmware

I just keep a browser window open to the web console and type "upgrade 1" (If 
I'm doing development, I probably want to see the logs anyway)

another way would be to have a bookmark to the upgrade page, at that point it's 
two clicks (three if you have to select the browser first) to do an ota upgrade

Is it really worth adding any code to the already tight firmware to save so 
little effort?

arendst, 2017-03-22:
I won't add as I do not have room for another 44k

closed 2017-03-27 by davidelang