arendst/Tasmota #259: Flash size 4MB (Sonoff Pow, Electrodragon…) -Tesp8266.flash.4m.ld

MacWyznawca, 2017-03-22:
I changed the settings in PlatformIO.ini (for Electrodragon) to 
Board = esp12e
-Tesp8266.flash.4m.ld

And now in Info I have:
Flash size:	4096kB
Program flash size: **4096kB**
Program size:	430kB
Free program space:	**592kB**

Earlier I had (with Board = esp01_1m and -Tesp8266.flash.1m64.ld):
Flash size:	4096kB
Program flash size:	**1024kB**
Program size:	429kB
Free program space:	**508kB**

Would anyone be so good and explain to me, where does the difference in memory available for the program and if something extra memory can come in handy?

arendst, 2017-03-22:
What do you think does the -T option do? It selects the linker to use and in your first case it links a 4096kB flash image while in the second it links the 1024kB flash image with 64k SPIFFS area. The ElectroDragon has 4M flash while (most) Sonoffs have just 1M flashes. I need the spiffs area to store my settings although I'm not using spiffs filemanager to handle it.

I wonder if your Electrodragon

MacWyznawca, 2017-03-22:
OKAY!
Thank you and I read more in the documentation.

From my tests: With the setting -Tesp8266.flash.4m.ld Electrodragon saves the settings in your firmware. I also checked the OTA update several times and it works in Electrodragon.

By the way: today I have to get new relays with ESP12f from Electrodragon (low voltage and multi-channel relay). How will I test to let you know what they are worth ;-)

closed 2017-03-22 by arendst

TRIROG, 2017-05-11:
@MacWyznawca 

> By the way: today I have to get new relays with ESP12f from Electrodragon (low voltage and multi-channel relay). How will I test to let you know what they are worth ;-)

Did you get the new relays? How do they perform?