arendst/Tasmota #260: Sonoff Dual

raisude, 2017-03-22:
In the Module parameters of the sonoff dual gpio4 is used to configure a sensor. Where can i find gpio4 on the board?

Regards Rainer

khcnz, 2017-03-22:
Use these resources
https://www.itead.cc/wiki/images/1/1d/Sonoff_Dual_View.pdf
http://tinkerman.cat/wp-content/uploads/2016/12/20161206_234331s.jpg
https://espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf

From that it appears its the unconnected pin on the bottom right of the ESP - i hope you have a fine soldering iron and you probably want to consider an external resister.

There wasn't an immediate google search that returned the result so it would be great if  you could capture on the wiki

Satellietje, 2017-03-22:
[image]

raisude, 2017-03-23:
Is this one wire as a bridge. Or are this two wires? And where can I connect the data wire of the sensor?

Regards Rainer

davidelang, 2017-03-23:
that would be the sensor data wire. Your sensor probably also needs 3.3v and ground connections (depending on what you are connecting)

raisude, 2017-03-23:
But the picture shows to solder points on the board. By looking at the picture I see one orange wire on the top and one yellow bellow. The yellow one ist the data connection? And what is the other one?

Satellietje, 2017-03-23:
The Orange is GPOI0 for start programming

Satellietje, 2017-03-23:
[image]

Lovely toys, the sonoff :-)

davidelang, 2017-03-27:
It appears that this issue has been solved, if there are other questions, please reopen this issue.

closed 2017-03-27 by davidelang