arendst/Tasmota #261: iTead 1 Channel Switch 5V / 12V Inching mode problem

SzilagyiR, 2017-03-22:
Hi!

I've upload firmware v4.0.8 in my iTead 1 Channel Switch 5V but It's not working properly.
The interlocking mode woking fine, but when I press the switch to change Inching mode, the relay continues on/off switching two times and then after a couple of seconds the switch freezes.

I changed Sonoff type in webserver to 1 Channel switch, but nothing happens.

Any solution?

Thanks: Robert

arendst, 2017-03-22:
Inching mode switch is not supported.

SzilagyiR, 2017-03-22:
I tried the http://mysonoffip/cm?cmnd=PulseTime%2011 for 1 sec pulse, and it works!

Thanks for your help!

closed 2017-03-27 by davidelang

jeylites, 2017-07-26:
Hi guys, how did you get the inching function going? Tried the link above but it's not working.

Is there inching support on the current Tasmota?

SzilagyiR, 2017-07-26:
No, but you can use pulsing mode, like me!

See wiki for more info!
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

In the URL, you must change mysonoffip to your sonoffs ip like 192.168.0.123.
It sets pulse mode for 1 sec, means it turne off after 1sec.

witek1308, 2017-08-11:
Hi, I've just configured for this method and I'm happy with how it works - thanks for the hint. At the same time - I have no idea (but it's pain for me), why after flashing to tasmota (5.5.2, maybe other releases too) the built in micro usb power doesn't work ? it used to work with original soft. the additional +5v for cables does but I'd prefer to use original micro usb cable rather than cut it. any hints? thanks in advance!

davidelang, 2017-08-11:
When you are flashing the ESP series of chips, it doesn't matter what firmware 
has been flashed before, that firmware isn't invoked. Putting the device into 
flash mode happens before any firmware code is executed.

The only problem you can have based on what's there before is if the new 
firmware doesn't overwrite some of the flash, and then uses that untouched flash 
for config info (which tasmota does do so that your config isn't lost when you 
do an update, it does have safety measures to try and avoid corruption)

So if you are having trouble getting it to flash, that has nothing to do with 
the tasmota firmware, and only has to do with your physical interface to the 
device, and the need to connect gpio0 to ground before power is applied.

If you connect gpio0 to ground before power is applied, the ESP family of chips 
_will_ go into flash programming mode, and there is nothing that you as a 
programmer can do to prevent this.

witek1308, 2017-08-11:
@davidelang thanks for your reply but I think we slightly misunderstood each other. I have no issue flashing my chip (done this), suprising is however that micro usb port stopped working after this operation, everything else works fine, I have full access to tasmota features. There is however problem with inching mode which I described here: [https://github.com/arendst/Sonoff-Tasmota/issues/453](url) where I have problem to make it reliable and inching. Maybe you can shed more light onto it?

davidelang, 2017-08-12:
Tasmota doesn't support inching mode, that is handled by a different chip on the 
board (at least for the old one, I assume the same for the new one)

witek1308, 2017-08-12:
Understood. Therefore I reached dead-end with Tasmota. Need to find out how to flash back the original firmware to have something

PeggyFree, 2017-08-12:
Maybe a warning about this limitation should be written in the wifi, in
"Devices - Additional Info"?
There may be a new page called something like "broken features" in order
to sum up.

witek1308, 2017-08-12:
I actually observed very interesting behaviour which gives me some hope which is:
==Setup
- I use PulseTime 5 to emulate inching
- my outer/2nd button is pressed to enable fix relay
== Behaviour
- when I disable MQTT in the configuration, relay behaves properly on power up - it DOESN'T trigger (at least after tens of tries it didn't) and it relays like inching
- when I enable MQTT it starts as described in my posts above - triggers on power up and behaves strange by flipping the relay

there must be something with MQTT... I tried MQTT PowerRetain 0 and 1 but it doesn't have anything to do. what I observed is just after enabling MQTT switch restarts and goes in this weird mode

senadaruc, 2018-07-01:
Gents, Is there a GPIO14 so I can install a magnetic sensor?

Thanks,