arendst/Tasmota #263: "Config Module" webinterface broken

CWempe, 2017-03-22:
After configuring my device as "Wemos D1 mini", I can define some sensors on the "Config Module" page.

But later this page does not load complete.

Here is an example:
[image]

Sometime it works after a reset.
But most of the times it doesn't.

Even wehen some drop down menus are shows the "save" button does not work.

I can reproduce this with a naked "Wemos D1 mini pro".

davidelang, 2017-03-22:
do any errors show up in the log?

do you have TLS enabled (we are running into a memory problem when TLS and the 
webserver are both enabled)

On Wed, 22 Mar 2017, Christoph Wempe wrote:

> After configuring my device as "Wemos D1 mini", I can define some sensors on the "Config Module" page.
>
> But later this page does not load complete.
>
> Here is an example:
> [image]
>
> Sometime it works [...]

arendst, 2017-03-22:
What browser are you using?

CWempe, 2017-03-22:
I did not check the logs yet. 🙈 
I just changed Syslog to "more debug".
But this is all I get during the broken page.

```
Mar 22 16:10:22 wemos02-7932 ESP-HTTP: Handle root
Mar 22 16:10:22 wemos02-7932 ESP-DSB: Sensor busy
Mar 22 16:10:23 wemos02-7932 ESP-HTTP: Handle config
Mar 22 16:10:24 wemos02-7932 ESP-Wifi: Checking connection...
Mar 22 16:10:24 wemos02-7932 ESP-Wifi: Connected
Mar 22 16:10:30 wemos02-7932 ESP-HTTP: Handle Module config
Mar 22 16:10:40 wemos02-7932 ESP-HTTP: Handle root
Mar 22 16:10:41 wemos02-7932 ESP-DSB: Sensor busy
Mar 22 16:10:43 wemos02-7932 ESP-DSB: Sensor busy
Mar 22 16:10:44 wemos02-7932 ESP-HTTP: Handle config
Mar 22 16:10:45 wemos02-7932 ESP-Wifi: Checking connection...
Mar 22 16:10:45 wemos02-7932 ESP-Wifi: Connected
Mar 22 16:10:49 wemos02-7932 ESP-HTTP: Handle Module config
Mar 22 16:11:06 wemos02-7932 ESP-Wifi: Checking connection...
Mar 22 16:11:06 wemos02-7932 ESP-Wifi: Connected
```

TLS is disabled.

I tried Chrome and Firefox.

arendst, 2017-03-22:
That log looks fine with me.

If it fails again look at the web page source. It could be big (18k+) but is should be one complete html page.

CWempe, 2017-03-22:
The browsers cannot show the source of the last page load, but need to load the page again.
So I cannot get the exact source for the currect page.

But in two attemps I got different source results.
It looks like the page (source) is not loeded completely. (I guess we knew that already)
The first attemt shows  no real HTML head and tail.
The second attemt had a valid HTML ending, but no correct start.

see:
```
ion value='7'>07 WS2812</option>
<option value='8'>08 IRremote</option>
<option value='9'>09 Switch1</option>
<option value='10'>10 Switch2</option>
<option value='11'>11 Switch3</option>
<option value='12'>12 Switch4</option>
<option value='13'>13 Button1</option>
<option value='14'>14 Button2</option>
<option value='15'>15 Button3</option>
<option value='16'>16 Button4</option>
<option value='17'>17 Relay1</option>
<option value='18'>18 Relay2</option>
<option value='19'>19 Relay3</option>
<option value='20'>20 Relay4</option>
<option value='21'>21 Relay1I</option>
<option value='22'>22 Relay2I</option>
<option value='23'>23 Relay3I</option>
<option value='24'>24 Relay4I</option>
<option value='25'>25 Led1</option>
<option value='26'>26 Led2</option>
<option value='27'>27 Led3</option>
<option value='28'>28 Led4</option>
<option value='29'>29 Led1I</option>
<option value='30'>30 Led2I</option>
<option value='31'>31 Led3I</option>
<option value='32'>32 Led4I</option>
<option value='33'>33 PWM1</option>
<option value='34'>34 PWM2</option>
<option value='35'>35 PWM3</option>
<option value='36'>36 PWM4</option>
<option value='37'>37 PWM5</option>
</select>
</br>
<br/>
<b>GPIO1</b> Serial Out<select id='g1' name='g1'>
	<option selected value='0'>00 None</option>
	<option value='1'>01 DHT11</option>
	<o='4'>04 DS18x20</option>
	<option value='5'>05 I2C SCL</option>
	<option value='6'>06 I2C SDA</option>
	<option value='7'>07 WS2812</option>
	<option value='8'>08 IRremote</option>
	<option value='9'>09 Switch1</option>
	<option value='10'>10 Switch2</option>
	<option value='11'>11 Switch3</option>
	<option value='12'>12 Switch4</option>
	<option value='13'>13 Button1</option>
	<option value='14'>14 Button2</option>
	<option value='15'>15 Button3</option>
	<option value='16'>16 Button4</option>
	<option value='17'>17 Relay1</option>
	<option value='18'>18 Relay2</option>
	<option value='19'>19 Relay3</option>
	<option value='20'>20 Relay4</option>
	<option value='21'>21 Relay1I</option>
	<option value='22'>22 Relay2I</option>
	<option value='23'>23 Relay3I</option>
	<option value='24'>24 Relay4I</option>
	<option value='25'>25 Led1</option>
	<option value='26'>26 Led2</option>
[... 124 lines cut ...]
		<option value='33'>33 PWM1</option>
		<option value='34'>34 PWM2</option>
		<option value='35'>35 PWM3</option>
		<option value='36'>36 PWM4</option>
		<option value='37'>37 PWM5</option>
	</select>
</br>
<br/>
<b>GPIO16</b>
<select id='g16' name='g16'>
	<option selected value='0'>00 None</option>
	<option value='1'>01 DHT11</option>
	<option value='2'>02 AM2301</option>
	<option value='3'>03 DHT22</option>
	<option value='4'>04></form>
</fieldset>
<br/>
<br/>
<form action='/cn' method='post'>
	<button>Configuration menu</button>
</form>
</div>
</body>
</html>
```

arendst, 2017-03-22:
What is the output of command status 0?

CWempe, 2017-03-22:
```
16:39:31 CMND: status 0
16:39:31 MQTT: stat/wemos02/STATUS = {"Status":{"Module":18, "FriendlyName":"Wemos02", "Topic":"wemos02", "ButtonTopic":"0", "Subtopic":"POWER", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
16:39:31 MQTT: stat/wemos02/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://htpc:80/arduino/wemos02.ino.bin", "Uptime":0, "Sleep":0, "BootCount":11, "SaveCount":22}}
16:39:31 MQTT: stat/wemos02/STATUS2 = {"StatusFWR":{"Program":"4.0.7", "BuildDateTime":"2017-03-21T23:52:59", "Boot":31, "Core":"00000000", "SDK":"2.0.0(656edbf)"}}
16:39:31 MQTT: stat/wemos02/STATUS3 = {"StatusLOG":{"Seriallog":2, "Weblog":2, "Syslog":4, "LogHost":"192.168.1.8", "SSId1":"CWempe", "SSId2":"indebuurt2", "TelePeriod":60}}
16:39:31 MQTT: stat/wemos02/STATUS4 = {"StatusMEM":{"ProgramSize":465, "Free":556, "Heap":17, "SpiffsStart":1024, "SpiffsSize":3052, "FlashSize":16384, "ProgramFlashSize":4096, "FlashChipMode":2}}
16:39:31 MQTT: stat/wemos02/STATUS5 = {"StatusNET":{"Host":"wemos02-7932", "IP":"192.168.1.43", "Gateway":"192.168.1.1", "Subnetmask":"255.255.255.0", "Mac":"5C:CF:7F:D9:5E:FC", "Webserver":2, "WifiConfig":3}}
16:39:31 MQTT: stat/wemos02/STATUS6 = {"StatusMQT":{"Host":"192.168.1.34", "Port":1883, "ClientMask":"DVES_%06X", "Client":"DVES_D95EFC", "User":"sonoffpow", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
16:39:31 MQTT: stat/wemos02/STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 22 15:39:31 2017", "Local":"Wed Mar 22 16:39:31 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
16:39:31 MQTT: stat/wemos02/STATUS10 = {"StatusSNS":{"Time":"2017-03-22T16:39:31"}}
16:39:31 MQTT: stat/wemos02/STATUS11 = {"StatusPWR":{"Time":"2017-03-22T16:39:31", "Uptime":0, "Vcc":3.026, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"CWempe", "RSSI":100}}}
```

arendst, 2017-03-22:
OK. You are using the not supported beta Arduino-ESP8266 library. This uses too much memory and makes rendering large pages imposibble.

Re-compile using the supported version 2.3.0 or use the pre-compiled versions please.

CWempe, 2017-03-22:
Ok.
I will try that later.
Thanks. 👍

closed 2017-03-22 by arendst

CWempe, 2017-03-22:
You were right.

The Webinterface seems to be stable now.
Thanks again!