arendst/Tasmota #265: FeatureRequest: Long Press for SONOFF TOUCH

lubeda, 2017-03-22:
I think that the Sonoff TOUCH is a true "interactive" device with human interaction and the other Sonoff modules are mostly controlled by Software. 

It would be nice if the TOUCH Module would be able to publish/handle two states:
- short press and 
- long press. 

For the short press the relay should toggle, i think this is the expected behavior.
For the long press it should be optional to toggle the relay, i think most user like to control this special event via software.

The long press should be not longer than 3 second and i think the "Pressing the button for over four seconds:" event has to be extended to 6 or more seconds.

Thanks for your great work

ionciubotaru, 2017-03-22:
Sonoff Touch respond to one click and double click already [https://github.com/arendst/Sonoff-Tasmota/issues/200](https://github.com/arendst/Sonoff-Tasmota/issues/200)

lubeda, 2017-03-22:
OK, in the log i can see the multipress 2 state but i dont know how to configure/set the ButtonTopic. I always get {"ButtonTopic":"0"} when i publish e.g. `cmnd/sonoff/buttontopic/Tralalala`  or `cmnd/sonoff_TouchBadOben/buttontopic/set  Tralalala`

davidelang, 2017-03-22:
just
/cmnd/sonoff/buttontopic tralalala

lubeda, 2017-03-23:
Perhaps my general understanding is wrong, but...
I expect on an singlepress  a toggle of the relay and a toggle of the MQTT Values POWER/RESULT, this work!!!
I expect on a double press a toggle of the relay and a some kind of new MQTT message with some kind of "tralalala" in it. This does not work take a look at this log:
`18:31:40 APP: Multipress 1
18:31:40 APP: Multipress 2
18:31:41 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
18:31:41 MQTT: stat/sonoff/POWER = OFF`

davidelang, 2017-03-23:
as noted in the other issue, a double press of the first button does one of two 
things

1. on a dual, it switches the second relay

2. on everything else, it switches the first relay

the stat/ messages are generated as a result of the relay being switched, not of 
the button being pressed.

ionciubotaru, 2017-03-23:
Open sonoff.ino and goto line 1352:

```
    case 2: // Toggle
      power ^= mask;
    }
    setRelay(power);

```
after `setRelay(power);` add:

```
if (multipress == 2 and sysCfg.module != SONOFF_DUAL) {
   mqtt_publish("cmnd/sonoff/buttontopic/Tralalala","0",false);
}
```

PeggyFree, 2017-09-01:
The long press is working now, maybe this ticket should be closed?

closed 2017-09-04 by lubeda