arendst/Tasmota #266: Webserver not reachable

raisude, 2017-03-22:
I have one sonoff basic and one sonoff dual. Both have the latest firmware. After some time both are not reachable via the webserver. But both are sending mqtt messages and are responding for mqtt requests like switching the relay. Is this normal behavior?

Regards Rainer

davidelang, 2017-03-22:
no, it's not normal behavior, what is your config? (do you have TLS enabled for 
example?)

can you send it a status command?

can you have it send logs somewhere?

raisude, 2017-03-23:
I have configured login in mode 4. But while I tried to connect  to the son off via browser there were only such messages every 20 seconds:

Mar 22 23:05:23 sonoff-7535 ESP-Wifi: Checking connection...
Mar 22 23:05:23 sonoff-7535 ESP-Wifi: Connected

davidelang, 2017-03-23:
what is your config (what module type, do you have TLS enabled, did you disable 
anything else)?

We know there are problems with the webserver running out of memory in many 
current configs.

On Thu, 23 Mar 2017, raisude wrote:

> I have configured login in mode 4. But while I tried to connect  to the son off via browser there were only such messages every 20 seconds:
>
> Mar 22 23:05:23 sonoff-7535 ESP-Wifi: Checking connection...
> Mar 22 23:05:23 sonoff-7535 ESP-Wifi: Connected
>
>

raisude, 2017-03-23:
I have enabled the DS18x20. The config is sonoff basic. And the second device sonoff dual.

Where can I look for if the TLS is enabled?

davidelang, 2017-03-23:
it's enabled in user_config*. It's disabled by default.

The basic and the dual are not expected to have any problems with the default 
config.

gemitro, 2017-03-24:
I also have two  Sonoffs basic configured in exactly the same way with static ip addresses. After some time (at least twice a day) only one of them is not accessible via the webserver.
When this device loses the webserver I can still ping it and send mqtt requests, as reboot.

raisude, 2017-03-24:
I have compiled the firmware with everything disabled not needed. The behaviour is the same as before. The Webserver is not generally dead. Sometimes the server responds and sometimes not. Not responding could be after power off the sonoff. Trying to connect to the server fails. Half an hour later without doing something I can connect.  5 minutes later clicking on a menu Point fails. Trying later could be success ... .

PaulBendall, 2017-03-24:
Might be having a similar issue although it is more around slowness than losing the webserver completely. For example I can send a HTTP command to my Sonoff PoW, http://_sonoff_ip_/cm?cmnd=Status%208 most of the time it will respond in less than 1 second, but sometimes it will take over 13 seconds. 

Another example is from the web portal clicking on "Information" will often fail on first attempt with page unavailable. Refreshing and the page is displayed

Program version : 4.0.8
Build Date & Time : 2017-03-22T18:31:58
Core/SDK version : 2_3_0/1.5.3(aec24ac9)

ESP Chip id : 10497513
Flash Chip id : 1458400
Flash size : 4096kB
Program flash size : 1024kB
Program size : 465kB
Free program space : 472kB
Free memory : 15kB

davidelang, 2017-03-24:
can you gather the logs (either serial or syslog) and see if anything shows up?

On Fri, 24 Mar 2017, raisude wrote:

> I have compiled the firmware with everything disabled not needed. The behaviour is the same as before. The Webserver is not generally dead. Sometimes the server responds and sometimes not. Not [...]

davidelang, 2017-03-24:
that could be it attempting to connect to a mqtt server. Please report what 
shows up in the logs (increase to 'more debugging' if nothing interesting shows 
up)
k

Nayar, 2017-03-24:
I flashed the lastest firmware on all my sonoffs: 4 Sonoff POW and 2 Sonoff classic. 

My sonoffs classic are not responding to HTTP or even ping. Sometimes some pings go.

But when i toggle the buttons, MQTT messages are sent immediately. But when I send MQTT messages to them, nothing. 

It seems like they are not liking "incoming connections"

davidelang, 2017-03-24:
both directions of mqtt are handled on the same connection, so it's not a 
network connection issue.

If you can get logs from it (syslog or serial are your best bets), we should be 
able to see what's happening.

Nayar, 2017-03-24:
[image]
[image]

Seems good

P.S. If you guys want graylog for your sonoffs, feel free to contact me.

davidelang, 2017-03-24:
Ok, failing to connect to the mqtt server can cause the sonoff to stall for 5 
sec at a time.

Nayar, 2017-03-24:
[image]

It tries to connect like every 5 mins according to my graph

davidelang, 2017-03-24:
can you increase the logging level and show a full cycle of logs (ideally as text not images)?

Nayar, 2017-03-24:
It responded to my MQTT message of setting logs to 4 instantly. 

```
"timestamp","message"
"2017-03-24T18:58:46.522Z","tele/SONOFF_1C1B81/STATE = {""Time"":""2017-03-24T18:58:45"", ""Uptime"":0, ""Vcc"":3.239, ""POWER"":""OFF"", ""Wifi"":{""AP"":1, ""SSID"":""iot.nayarweb.com"", ""RSSI"":92}}"
"2017-03-24T19:02:31.396Z","{""Time"":""2017-03-24T19:02:30"", ""Uptime"":1}"
"2017-03-24T19:03:48.179Z","{""Time"":""2017-03-24T19:03:47"", ""Uptime"":1, ""Vcc"":3.236, ""POWER"":""OFF"", ""Wifi"":{""AP"":1, ""SSID"":""iot.nayarweb.com"", ""RSSI"":94}}"
"2017-03-24T19:04:34.860Z","Attempting connection..."
"2017-03-24T19:04:35.865Z","Query done with 0 mqtt services found"
"2017-03-24T19:04:36.757Z","Connected"
"2017-03-24T19:04:37.224Z","Online (retained)"
"2017-03-24T19:04:37.641Z","cmnd/SONOFF_1C1B81/POWER ="
"2017-03-24T19:08:54.633Z","{""Time"":""2017-03-24T19:08:52"", ""Uptime"":1, ""Vcc"":3.238, ""POWER"":""OFF"", ""Wifi"":{""AP"":1, ""SSID"":""iot.nayarweb.com"", ""RSSI"":96}}"
"2017-03-24T19:08:55.050Z","{""SysLog"":""4 (Active 4)""}"
"2017-03-24T19:08:55.054Z","Receive topic cmnd/SONOFF_1C1B81/SysLog, data size 1, data 4"
"2017-03-24T19:08:55.054Z","DataCb Topic SONOFF_1C1B81, Group 0, Index 1, Type SYSLOG, Data 4 (4)"
"2017-03-24T19:08:55.516Z","{""SysLog"":""4 (Active 4)""}"
"2017-03-24T19:08:55.750Z","Saved configuration (1348 bytes) to flash at F7 and count 1544"
"2017-03-24T19:09:04.915Z","Checking connection..."
"2017-03-24T19:09:04.916Z","Connected"
"2017-03-24T19:09:24.940Z","Checking connection..."
"2017-03-24T19:09:24.941Z","Connected"
"2017-03-24T19:09:44.963Z","Connected"
"2017-03-24T19:09:44.960Z","Checking connection..."
"2017-03-24T19:10:04.918Z","Checking connection..."
"2017-03-24T19:10:04.930Z","Connected"
"2017-03-24T19:10:24.915Z","Checking connection..."
"2017-03-24T19:10:24.915Z","Connected"
"2017-03-24T19:10:44.916Z","Checking connection..."
"2017-03-24T19:10:44.919Z","Connected"
"2017-03-24T19:11:05.272Z","Checking connection..."
"2017-03-24T19:11:05.338Z","Connected"
"2017-03-24T19:11:25.274Z","Checking connection..."
"2017-03-24T19:11:25.274Z","Connected"
```

It's continuing like this while cannot ping.

davidelang, 2017-03-24:
On Fri, 24 Mar 2017, Nayar Joolfoo wrote:

> "timestamp","message"

Ok, there is some missing info here (probably called programname or syslogtag) 
that would tell us which subsystem is generating the message.

But I'll bet that the checking connection, connected messages are from the wifi, 
and that would show it having problems about ever 20 seconds

the fact that you cannot ping it while it is on the network is odd. can you make 
sure you aren't trying to do DNS lookups while pinging (on linux/osx this is 
adding -n to the ping command)

> "2017-03-24T18:58:46.522Z","tele/SONOFF_1C1B81/STATE = {""Time"":""2017-03-24T18:58:45"", ""Uptime"":0, ""Vcc"":3.239, ""POWER"":""OFF"", ""Wifi"":{""AP"":1, ""SSID"":""iot.nayarweb.com"", [...]

tam481, 2017-03-24:
Having a similar issue with Sonoff Basic. No TLS (yet) or anything else other than logging turned off and DHT22 connected to GPIO14.

Firmware version 4.0.4

I thought it was my router until I read this issue and tried sending an MQTT command and it worked fine while the web server is still not connecting.

Nayar, 2017-03-24:
```
"2017-03-24T19:08:55.050Z",,"SONOFF_1C1B81-7041 ESP-MQTT: stat/SONOFF_1C1B81/RESULT = {""SysLog"":""4 (Active 4)""}","{""SysLog"":""4 (Active 4)""}"
"2017-03-24T19:08:55.054Z",,"SONOFF_1C1B81-7041 ESP-RSLT: Receive topic cmnd/SONOFF_1C1B81/SysLog, data size 1, data 4","Receive topic cmnd/SONOFF_1C1B81/SysLog, data size 1, data 4"
"2017-03-24T19:08:55.054Z",,"SONOFF_1C1B81-7041 ESP-RSLT: DataCb Topic SONOFF_1C1B81, Group 0, Index 1, Type SYSLOG, Data 4 (4)","DataCb Topic SONOFF_1C1B81, Group 0, Index 1, Type SYSLOG, Data 4 (4)"
"2017-03-24T19:08:55.516Z",,"SONOFF_1C1B81-7041 ESP-MQTT: stat/SONOFF_1C1B81/RESULT = {""SysLog"":""4 (Active 4)""}","{""SysLog"":""4 (Active 4)""}"
"2017-03-24T19:08:55.750Z",,"SONOFF_1C1B81-7041 ESP-Config: Saved configuration (1348 bytes) to flash at F7 and count 1544","Saved configuration (1348 bytes) to flash at F7 and count 1544"
"2017-03-24T19:09:04.915Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:09:04.916Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
"2017-03-24T19:09:24.940Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:09:24.941Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
"2017-03-24T19:09:44.963Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
"2017-03-24T19:09:44.960Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:10:04.918Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:10:04.930Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
"2017-03-24T19:10:24.915Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:10:24.915Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
"2017-03-24T19:10:44.916Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:10:44.919Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
"2017-03-24T19:11:05.272Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:11:05.338Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
"2017-03-24T19:11:25.274Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:11:25.274Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
"2017-03-24T19:11:45.272Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:11:45.272Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
"2017-03-24T19:12:05.303Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:12:05.304Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
"2017-03-24T19:12:25.275Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:12:25.275Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
"2017-03-24T19:12:45.277Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:12:45.277Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
"2017-03-24T19:12:55.473Z",,"SONOFF_1C1B81-7041 ESP-MQTT: Attempting connection...","Attempting connection..."
"2017-03-24T19:12:56.477Z",,"SONOFF_1C1B81-7041 ESP-mDNS: Query done with 0 mqtt services found","Query done with 0 mqtt services found"
"2017-03-24T19:12:57.310Z",,"SONOFF_1C1B81-7041 ESP-MQTT: Connected","Connected"
"2017-03-24T19:12:57.768Z",,"SONOFF_1C1B81-7041 ESP-MQTT: tele/SONOFF_1C1B81/LWT = Online (retained)","Online (retained)"
"2017-03-24T19:12:58.186Z",,"SONOFF_1C1B81-7041 ESP-MQTT: cmnd/SONOFF_1C1B81/POWER =","cmnd/SONOFF_1C1B81/POWER ="
"2017-03-24T19:13:09.177Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Checking connection...","Checking connection..."
"2017-03-24T19:13:09.204Z",,"SONOFF_1C1B81-7041 ESP-Wifi: Connected","Connected"
```

Updated the logs with full message

Nayar, 2017-03-24:
Filtering Wifi module from logs, I get this during last 30 mins

```
"timestamp","full_message","message","tasmota_module"
"2017-03-24T19:08:55.050Z","SONOFF_1C1B81-7041 ESP-MQTT: stat/SONOFF_1C1B81/RESULT = {""SysLog"":""4 (Active 4)""}","{""SysLog"":""4 (Active 4)""}","MQTT"
"2017-03-24T19:08:55.054Z","SONOFF_1C1B81-7041 ESP-RSLT: Receive topic cmnd/SONOFF_1C1B81/SysLog, data size 1, data 4","Receive topic cmnd/SONOFF_1C1B81/SysLog, data size 1, data 4","RSLT"
"2017-03-24T19:08:55.054Z","SONOFF_1C1B81-7041 ESP-RSLT: DataCb Topic SONOFF_1C1B81, Group 0, Index 1, Type SYSLOG, Data 4 (4)","DataCb Topic SONOFF_1C1B81, Group 0, Index 1, Type SYSLOG, Data 4 (4)","RSLT"
"2017-03-24T19:08:55.516Z","SONOFF_1C1B81-7041 ESP-MQTT: stat/SONOFF_1C1B81/RESULT = {""SysLog"":""4 (Active 4)""}","{""SysLog"":""4 (Active 4)""}","MQTT"
"2017-03-24T19:08:55.750Z","SONOFF_1C1B81-7041 ESP-Config: Saved configuration (1348 bytes) to flash at F7 and count 1544","Saved configuration (1348 bytes) to flash at F7 and count 1544","Config"
"2017-03-24T19:12:55.473Z","SONOFF_1C1B81-7041 ESP-MQTT: Attempting connection...","Attempting connection...","MQTT"
"2017-03-24T19:12:56.477Z","SONOFF_1C1B81-7041 ESP-mDNS: Query done with 0 mqtt services found","Query done with 0 mqtt services found","mDNS"
"2017-03-24T19:12:57.310Z","SONOFF_1C1B81-7041 ESP-MQTT: Connected","Connected","MQTT"
"2017-03-24T19:12:57.768Z","SONOFF_1C1B81-7041 ESP-MQTT: tele/SONOFF_1C1B81/LWT = Online (retained)","Online (retained)","MQTT"
"2017-03-24T19:12:58.186Z","SONOFF_1C1B81-7041 ESP-MQTT: cmnd/SONOFF_1C1B81/POWER =","cmnd/SONOFF_1C1B81/POWER =","MQTT"
"2017-03-24T19:25:03.235Z","SONOFF_1C1B81-7041 ESP-MQTT: Connected","Connected","MQTT"
"2017-03-24T19:25:03.738Z","SONOFF_1C1B81-7041 ESP-MQTT: tele/SONOFF_1C1B81/LWT = Online (retained)","Online (retained)","MQTT"
"2017-03-24T19:25:04.154Z","SONOFF_1C1B81-7041 ESP-MQTT: cmnd/SONOFF_1C1B81/POWER =","cmnd/SONOFF_1C1B81/POWER =","MQTT"
"2017-03-24T19:24:59.726Z","SONOFF_1C1B81-7041 ESP-MQTT: Attempting connection...","Attempting connection...","MQTT"
"2017-03-24T19:25:00.733Z","SONOFF_1C1B81-7041 ESP-mDNS: Query done with 0 mqtt services found","Query done with 0 mqtt services found","mDNS"

```

davidelang, 2017-03-27:
there were significant improvements in 4.1.0, can you check to see if you still have problems with it?

Nayar, 2017-03-28:
After flashing the device to 4.1.0 via serial, the sonoff is pingable and web server accessible. 

Lots of packets loss though: 12% despite the sonoff is within 2 m from my AP

davidelang, 2017-03-28:
well, that's not going to be a software problem, we have seen reports from 
people that found bad solder joints on their modules, in some cases making them 
unable to connect to wifi at all when powered by mains power.

try powering the device from your serial cable and see if you have the same 
problem or not.
k

closed 2017-04-02 by davidelang