arendst/Tasmota #267: Wifi not working on mains

gid204, 2017-03-22:
When I have my sonoff basic hooked up to serial port via pc it connects and is completely reachable on the network. However when I disconnect serial and connect via mains wifi fails to connect to my home network. I try to reconfigure with 4 button presses but the wifi configuration doesn't even pop up.

Any help is greatly appreciated :)

davidelang, 2017-03-23:
we've seen some devices with cold solder joints in them. If there is a bad joint 
it could either not provide proper power, or generate so much noise that it 
affects wifi

gid204, 2017-03-23:
Ok thanks for that, I'll try it out again with one of the other ones I have otherwise I'll get some new ones  ordered. thanks for your help :)

smadds, 2017-03-23:
Does the LED light up OK & can you hear the relay click when you single press the button?

gid204, 2017-03-23:
Yes led lights up just fine.
On serial connection it lights up and then connects to network

On mains it lights up and then just flashes as it can't connect to network. I try to force it with 4 button press but it just doesn't react to it. Button works though because when I press it the light turns on and off...

citrusfizz, 2017-03-24:
thats interesting, if the relay is working, the ESP8266 is working. 
Can you check any logs on your Accesspoint to see if it try to connect?

davidelang, 2017-03-27:
any further info on this issue?

gid204, 2017-03-28:
Unfortunately not..been busy with uni last few days. Going to try troubleshoot a few more things then will opt for purchasing replacements.

These units were bought off eBay (still came in the official box). I'll get them off itead site this time..

Shaewen-Chronicles, 2017-03-29:
Ich habe the exact same issue in the WiFi log it says the Device is connecting and then lossing the connection but if i have it sitting at the same place with seriell connection its working fine.

ekgerke, 2017-04-01:
I had an issue with wifi dropping with one module. Turned out that I had a defective LED bulb that somehow interfered with wifi. Replacing the bulb ended the problem for good.

Shaewen-Chronicles, 2017-04-01:
Okay i will trie it with the bulb removed

Shaewen-Chronicles, 2017-04-01:
No even with the bulb removed it gets no Wifi connection

davidelang, 2017-04-02:
did you check for bad solder connections on the device?

Shaewen-Chronicles, 2017-04-03:
I think i will do that tomorrow

gid204, 2017-04-06:
my new units came (directly from itead and not eBay).
They seem to be working perfectly now :)

closed 2017-04-23 by davidelang

boko6, 2017-06-10:
Hi, I have exactly the same issue, as in issue 391, the only difference with the rest of sonoffs that i have is that this one was working and linked with the ewlink app

glsf91, 2017-11-05:
I had the same problem with a Sonoff basic but solved (I hope). See my reply in #391