arendst/Tasmota #268: Trigger a tele publish

smadds, 2017-03-23:
I may have missed something, but is there any way to trigger a "tele" publish from a device using a command? 

I'm using node-red and dashboards can take a while to draw when the dashboard page is first opened. If I could trigger the periodic tele publish using an MQTT command it would be very helpful.

davidelang, 2017-03-23:
no, tele/ reports are only generated at a periodic interval. You can set this 
interval pretty small, but I've run into trouble setting it below 5-10 sec

David Lang

smadds, 2017-03-23:
In that can, can I make an enhancement request for a sendtele command (no payload) to manually trigger sending a tele? This would allow the normal period to be longer if it could be requested on demand.

arendst, 2017-03-23:
There is more to it. Some sensors require setup time before providing their information. That's what I do a second before sending the telemetry data.

I'll might look into it...

davidelang, 2017-03-27:
I'm working on adding support for the wion power sensor module, and it has the interesting property that every time you read the sensor, it resets the data, so issuing extra tele/ statements would have 'interesting' results on the numbers.

tovtoc, 2017-05-12:
Yo can trigger a response issuing a a Status command.  I'm using it for SonoffPow with MQTT
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow.
In that case you should publish "cmnd/Sonoff-ABCD/status -m 0" -m1 would trigger a different response 
And expect the result in "stat/Sonoff-ABCD/STATUS8", in my case I'm looking for power data. The JSON is formatted differently to "tele/" response so you look for the right path for the values you are interested.

My problem now is that response it's sent immediately and does not wait for the power measure to stabilize, I trigger the response with the switch turn on command. So I'm looking to delay the answer a few ms to get the appropriate response from Sonoff module.

ardeus-ua, 2018-06-08:
Here's how I am doing it. 
I have Home Assistant that needs to wait for telemetry to get temperature/humidity/power information for each Tasmota device. Sometimes it takes up to 5 minutes since all of them have 300 seconds for teleperiod. This caused ventilation to switch on based on comparison to null value. I have tried to rewrite rules to check if value is not null, but found a simpler solution. 
I am simply sending payload "300" to "cmnd/sonoffs/teleperiod". As a response I get that teleperiod is 300 sec's and usual telemetry message

ascillato2, 2018-07-10:
Hi,

This issue seems to be addressed. Closing. Please, reopen if needed

closed 2018-07-10 by ascillato2

diepes, 2019-05-30:
I came here looking for a way to trigger a tele/xxx/SENSOR update, so i dont have to wait for the 5min.

@ardeus-us 's workaround worked sending cmnd/xxx/teleperiod "300"

This feels a bit like a hack, should we not have it that sending to cmnd/xxxx/SENSOR   and that triggers a response back tele/xxxx/SENSOR ?

My use case is a mqtt app, and i just need it when i need a quick update, just after connecting and got some old data displayed.  Would the other option be to change the QOS for the tele/xxx/SENSOR data ?

meingraham, 2019-05-30:
@diepes - `Status 8`?

Please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)