arendst/Tasmota #269: FeatureRequest

ardeus-ua, 2017-03-23:
Please add AM2320 temparature/humidity sensor

gemitro, 2017-03-23:
I have exactly the same request.

davidelang, 2017-03-23:
Can you please point at information on this sensor and what is needed to talk to 
it? The more info that is provided (the less digging that someone adding the 
support needs to do), the easier, and therefor more likely it is to add the 
support.

altelch, 2017-03-23:
Looks like this one: https://github.com/thakshak/AM2320

Greetings,

           Heiko.

arendst, 2017-03-23:
I just read through this one https://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html and the AM2320 is not among the most desired ones:

> In summary, they appeared similar to the DHT22 most of the time, but I had so many different intermittent problems I would not intend to use these devices in future experiments.

ardeus-ua, 2017-03-23:
I've been checking with one sensor on previous releases of Sonoff-MQTT-OTA and it worked like a charm with AM2302 option configured. Once I have updated my sonoff basic to Tasmota it is not recognised no matter which option on GPIO14 (where it's connected through 4k7 resistor between + and GPIO14 as per specifications of the sensor).

arendst, 2017-03-25:
So it's a AM2302 and not a AM2320.

ardeus-ua, 2017-03-25:
I can definitely say that it says AM2320 on it

arendst, 2017-03-26:
Ok. the AM2320 can be connected as a AM2302 or via I2c but I do not support it on I2c. Glad it works for you.

closed 2017-03-26 by arendst

ardeus-ua, 2017-03-26:
It is not working. it WAS working. It stopped working on Tasmota.