arendst/Tasmota #272: AP Mode when no WiFi

tombartosz, 2017-03-23:
I would like to see feature like this.
When first and second wi-fi connection fails and web username and password is set - SonOff goes into AP mode.

It can be useful in situation like this:
I have SonOff connected to the light. It is hard to reach regular switch and sometimes wi-fi is off. In this case I have to enable HotSpot mode in my smartphone (of course I have HotSpot name and password configured as second wi-fi in SonOff) and then I can turn on light using smartphone. 

This feature I'm proposing would allow me just to connect to SonOff AP and turn on/off switch.

davidelang, 2017-03-23:
That's what WIFI_MANAGER (option 2) does.

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

WifiConfig     |           | Show current config tool
WifiConfig     | 0         | Disable wifi config but restart (used with alternate AP)
WifiConfig     | 1         | Start smart config for 1 minute and set as current config tool
WifiConfig     | 2         | Start wifi manager (web server at 192.168.4.1) and set as current config tool
WifiConfig     | 3         | Start WPS config for 1 minute and set as current config tool
WifiConfig     | 4         | Disable wifi config but retry other AP without restart

it only keeps the AP mode active for 2 min (since there's no password on it, 
anyone could connect and start reconfiguring the device)

tombartosz, 2017-03-23:
Yes, I know that I can put device in AP mode (but I need connection or press a button), but I would like device to go into AP mode automatically when no WiFi connection. Another (worst) option is to allow AP mode to be active forever and add username and password.

arendst, 2017-03-23:
Have you ever tried it? It should do what you expect it to do.

davidelang, 2017-03-23:
set WIFI_CONFIG_TOOL = WIFI_MANAGER in user_config.h and it will come up in AP 
mode when it powers on.

arendst, 2017-03-23:
Or just execute command wificonfig 2 and enjoy it's feature

rhodopsik, 2017-03-23:
Monday I also tried this, I set WIFI_CONFIG_TOOL = WIFI_MANAGER, but at my device it always worked in WPS Mode.
I only mention this if this could be a bug?

davidelang, 2017-03-27:
the AP mode is only up for 2 minutes, could that be confusing you?

rhodopsik, 2017-03-28:
No, that was the serial output, after flashing with different parameters...

davidelang, 2017-03-28:
logs please.

gonzzovela, 2017-04-14:
I also would like to see as an option to configure SonOff as AP and connect with the mobile to turn on/off the switch

Setting WIFI_MANAGER is not viable as it does not allow to toggle the switch. Furthermore, I'd okie to maintain the AP mode and not change to STA after 2 mins @

davidelang, 2017-04-23:
@gonzzovela see #212 about the timeout question on the ap mode. Please move discussion of AP mode remaining active there.

I'm going to close this issue as there is no other feedback, please re-open if there is more troubleshooting of this issue.

closed 2017-04-23 by davidelang