arendst/Tasmota #274: Time schedule function

gthb96, 2017-03-23:
Hello,
Is it possible that there will be a time schedule function on the WebIF so we can set a specific time when it switches?

davidelang, 2017-03-23:
It's not likely, that sort of logic should be implemented in your home 
automation software that the devices connect to.

Keep in mind that these devices do not have a reliable clock, they need to set 
their clock from the network, so if you are looking for something that doesn't 
depend on the network being available, time based changes wouldn't be reliable 
either.

davidelang, 2017-03-25:
hijacking this a little bit

@arendst what are you meaning by the 'help wanted' tag here? ususally "help wanted" means that we are waiting for more info from the reporter (I've thought about asking for a "waiting for info" tag to use instead to make it clearer what needs to happen before anything more can be done), but that doesn't seem to be the case here.

arendst, 2017-03-25:
I use help wanted to indicate to the community to provide info to the question.

davidelang, 2017-04-23:
closing this issue as it is not something that can be done reliably by the firmware in the face of power outages. please re-open if you think this should receive further work.

closed 2017-04-23 by davidelang