arendst/Tasmota #275: Is there anyway that can clear setting on ESP-07?

hamwong, 2017-03-23:
I am wonder is there anyway that I can clear the setting if error setting make the ESP in infinity bootup loop ?
Also how to use ADC with Tasmota firmware?

By the way, ESP-07 works very good with sonoff, I connect 1xDHT11, 1xBMP180, 1xBH1750, 1xHC-SR501.
I still got 3xGPIO, 1xADC and serial Tx Rx left. it works good,
[image]

davidelang, 2017-03-23:
If it stays up long enough, you can hold the button down for 4 seconds to 
overwrite the saved config with what was compiled in by user_config*

If not, change the value of CFG_HOLDER in user_config* and flash it. That will 
decide that the saved config is invalid and overwrite it.

hamwong, 2017-03-23:
thanks davide
do you have any hints for me regarding ADC? how to use this pin?

davidelang, 2017-03-23:
what sort of thing are you trying to use it for?

hamwong, 2017-03-24:
I am thinking using something like soil sensor or rain detector there is a ADC, no extra ADC require for that.
I wish MQ-2 Gas Sensor and HY-SRF04 / 05 will soon add and support in Tasmota

khcnz, 2017-03-24:
It was added recently. You need to compile a special build that enables the ADC 
look for the #def USE_ADC_VCC for more info

hamwong, 2017-03-24:
I have activated #define USE_ADC_VCC            1    
in sonoff_template.h, I have GPIO_ADC0 at last line
I use light swtich with Analog out, I don't see any difference on VCC (in case it is the value) and it keep on same lavel.

00:59:51 MQTT: tele/ESP07-MultiSensor-1/STATE = {"Time":"2017-03-25T00:59:51", "Uptime":0, "Vcc":4.094, "Wifi":{"AP":1, "SSID":"Ham2", "RSSI":90}}
00:59:51 MQTT: tele/ESP07-MultiSensor-1/SENSOR = {"Time":"2017-03-25T00:59:51", "Switch1":"ON", "DHT11":{"Temperature":26.0, "Humidity":55.0}, "BMP180":{"Temperature":26.9, "Pressure":1015.3}, "BH1750":{"Illuminance":1}}

I don't found ADC value

@arendst A little Hint will be apperciated

arendst, 2017-03-25:
If you want ADC input you'll have to disable USE_ADC_VCC (//define USE_ADC_VCC) in user_config.h. Then you see the input value in status 10 and telemetry.

hamwong, 2017-03-25:
I just did, I just look at the code and discover it,
I found no matter I connect ADC pin or not, result is 1024
02:25:50 MQTT: tele/ESP07-MultiSensor-1/SENSOR = {"Time":"2017-03-26T02:25:50", "Switch1":"ON", "AnalogInput0":1024, "DHT11":{"Temperature":22.0, "Humidity":43.0}, "BMP180":{"Temperature":24.4, "Pressure":1019.0}, "BH1750":{"Illuminance":9}}

citrusfizz, 2017-03-30:
remember that the esp adc pin is like 1.1v input i think.  anything over that is going to show 1024
i have not personally tested this with this software however. 

personally i think the ADC pin on this thing is junk  i tried a couple things and with out implementing a voltage divider with the right math to get it the input voltage to be between 0v and 1.1v  its pretty much is useless. 

I could be talking out of my ass however, i haven't done much except a few failed projects trying to use that pin on the ESP8266 (again, not using this software)

hamwong, 2017-03-31:
well as I only try to use it as experimental, its OK if this not work, I just curious how does ADC works, 
as ESP command is same using analogRead(A0) either here or on other sketch, I don't think problem is on Tasmota, I just wonder why my 2 ESP-07 all reading 1024  even there is nothing connect to that pin? if anyone else is getting value other than me, than I think it may due to my bad soldering skill.

P.S. after long run on ESP-07 and I put it in a waterproof box, analog input  drop to 698 once than back to 1024 on next tele

citrusfizz, 2017-04-01:
Well,  the ADC pin might be high by default.  i know other pins on my nodemcu module are   to use a button i had to write code to detect when it was pulled low.   not sure how that works with the ADC pin tho. 
if its high default with no wiring,  then 1024 response seems right. 
but again, i'm not sure

davidelang, 2017-04-02:
with nothing driving it, the adc is going to float, and since it reads the voltage between soemthing like 0-1.2v not 0-3.3v, most of it's range in floating is going to be above the 1024 level

closed 2017-04-02 by davidelang

hamwong, 2017-04-02:
hi @davidelang,

can you say a bit more, what do you mean  by "it's range in floating is going to be above the 1024 level"?
adc value is 0-1024, if value will always/mostly be return at 1024 either connect to a analog pin or unconnected, than what is it for? and why value is going to above 1024?

davidelang, 2017-04-02:
if it's not connected to anything, it's voltage will drift between 0 and 3.3 
volts randomly.

but as I understand it, the adc converts values between 0 and 1.2v to numbers 
0-1024, so any time that it reads above 1.2v, it's going to read 1024.

If it's allowed to drift randomly, it will read 1024 the majority of the time.

to use it, you need to connect the adc pin to something that has a voltage 
between 0 and 1.2v, either from a source in that range, or via some sort of 
voltage divider (like a resister network) or through a transformer, etc

hamwong, 2017-04-02:
thanks for detail expalin @davidelang 
Have a nice day, I will try to check further with ADC in spare time.

ngalfas, 2017-12-04:
This is worth of further exploit since it can support the use of many more sensors for the Sonoff/Tasmota combo in home automation scenarios.
Ie. i saw no projects with  MQ-xx  gas sensors or  smoke-dust sensors for security purposes on a more complete scenario.
What i mean is to add some new wiki pages specific to these sensors and exact instuctions.