arendst/Tasmota #276: upload error platform.io

amandus99, 2017-03-23:
Hi I just downloaded everything but I can't upload a firmware. I press the switch before connecting the sonoff with the pprogrammer to the macbook. Compiling the firmware works but uploading is failing after about 11seconds. Any idea?

`text       data     bss     dec     hex filename
460273    10096   34008  504377   7b239 .pioenvs/sonoff/firmware.elf
Building .pioenvs/sonoff/firmware.bin
Looking for upload port...
Auto-detected: /dev/cu.usbserial-00000000
Uploading .pioenvs/sonoff/firmware.bin
warning: espcomm_sync failed

error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 255
========================== [ERROR] Took 11.18 seconds ==========================`

davidelang, 2017-03-23:
that indicates that you aren't connecting to the device, either you have 
something else providing power to it (you hvae to hit the button before 
providing power to the device), or your serial pinout isn't correct.

is the device lighting up when you connect it to the programmer?

try swapping the tx/rx pins, it's common to have these wrong.

k

amandus99, 2017-03-23:
I tried to switch rx/tx but it was right. RX is conected to TX and vice versa. But still no difference. What do you mean by something else providing power to it? Is the 3.3V output of the programmer not enough?

davidelang, 2017-03-23:
the power from the programmer should be enough, I was making sure you didn't 
have something else providing power as well.

double check that you are using the correct serial port
kk

amandus99, 2017-03-23:
there is only 

/dev/cu.Bluetooth-Incoming-Port	/dev/cu.usbserial-00000000

so I guess the autodetect is doing right.

How can I check that the ESP is started in flashmode?

davidelang, 2017-03-23:
the led does something different than if you boot up normally (IIRC it blinks)
k

amandus99, 2017-03-23:
oh that's weired...
So I took my second sonoff I bought and tried with this and with this device it is working properly. The only difference is that this isn't a RF version. Any idea what could be the problem with the RF version?

amandus99, 2017-03-23:
ok there is really a differnece. Sorry for not reading properly. After the trick with the screwdriver it works as well. Can be closed. Can I close the issue by myself?

closed 2017-03-23 by amandus99

jkerdsri, 2017-09-11:
What is your trick then? please share, I have the same problem.