arendst/Tasmota #278: power on state not working?

amandus99, 2017-03-23:
#define APP_POWERON_STATE      1                 // [PowerOnState] Power On Relay state (0 = Off, 1 = On, 2 = Toggle Saved state, 3 = Saved state)

I set APP_POWERON_STATE to 1 this I would expect that the LED is always on if I connect to power but it seems that it is always the last state that is started. But would be 3?

amandus99, 2017-03-23:
I flashed the sonoff and the sonoff RF with the same firmware. On the RF the behavior is like expected on the "normal" sonoff it isn't ??

knotbeerdan, 2017-04-09:
I am having the same issue on my sonoff basic module.

I have tried with and without the retain settings below
#define MQTT_BUTTON_RETAIN     0
#define MQTT_POWER_RETAIN      0
#define MQTT_SWITCH_RETAIN     0

I have also tried all 4 settings for the line below
#define APP_POWERON_STATE      1

Those are my current settings and the units do not switch the relay on, even before it connects to the MQTT server.

davidelang, 2017-04-23:
have you tried setting the values in the config of the device? see the troubleshooting page for why changes to the user_config.h may not be getting applied to the device

knotbeerdan, 2017-04-24:
I was able to get it to work as expected, thank you!. One question I have is what are you supposed to change the value of CFG_HOLDER to? I am sorry if this is a dumb question as I am not at any type of "advanced" level of programming and it seems that references a memory location or something alike.

Also, is there a way to stop it from turning off momentarily when it connects to the MQTT server? I am assuming that is when it shuts opens the relay then closes it again. The only reason I ask is that mine is connected to LED bulbs, so there is an additional  longer delay before the bulbs turn on.

davidelang, 2017-04-24:
the cfg_holder value does not point to anything, it is only compared to the 
value in a specific memory location, if they match, it assumes the existing 
config is valid, if it doesn't, it assumes the configs are corrupt.

unless the mqtt server instructs the sonoff to switch, connecting to one will 
not switch the relays by themselves. check your logs (possibly increasing the 
logging level to see more details) and see what's really happening.
k

closed 2018-05-08 by stale[bot]