arendst/Tasmota #279: Build firmware.bin

SzilagyiR, 2017-03-24:
HI!

How can I create custom firmware.bin with edited user_config.h file?
I want to update firmware over webserver.

Thanks: Robert

davidelang, 2017-03-24:
Take a look at the wiki. Read through the "Getting Started" section. It will 
probably answer your questions.

smadds, 2017-03-24:
You can use the Arduino IDE: _Sketch, Export Compiled Binary_.
It will compile & save the binary image.

SzilagyiR, 2017-03-24:
Thanks, I missed it!

closed 2017-03-25 by arendst

SzilagyiR, 2017-03-27:
When I try it, I gave this error:

[image]

davidelang, 2017-03-27:
look at the getting started section in the wiki, you are missing libraries that you need to install

SzilagyiR, 2017-03-27:
Thanks, I look after it!

SzilagyiR, 2017-03-28:
It works, thanks!

seetendra, 2018-10-03:
Just a question on this . when we export the binary file do we know where is it stored

andrethomas, 2018-10-03:
> Just a question on this . when we export the binary file do we know where is it stored

Usually for Arduino (as seen in the cases above) the exported bin file will be in the same folder as the code you are compiling.