arendst/Tasmota #281: How to get State for Switch Topic?

prasad7676, 2017-03-25:
I have Main topic is...sonoff
switchtopic is.....switch

how to get stat for switch topic ... i tried Status 10 but its always says Switch1 OFF

[image]
[image]

davidelang, 2017-03-25:
stat/ happens when a relay is changed. Switches are not relays. You will get a 
cmnd/switch/power message not a status message

status 10 shows if the switch is open or closed at the time the status command 
is issued.

On Fri, 24 Mar 2017, prasad7676 wrote:

> Date: Fri, 24 Mar 2017 17:14:13 -0700
> From: prasad7676 <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f809802fe7c2437fa7e503683f07c23a30535608792cf0000000114ed785
>     592a169ce0cee2497@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: [arendst/Sonoff-Tasmota] How to get State for Switch Topic? (#281)
> 
> I have Main topic is...sonoff
> switchtopic is.....switch
>
> how to get stat for switch topic ... i tried Status 10 but its always says Switch1 OFF
>
> [image]
> [image]
>
>
>

davidelang, 2017-03-27:
did the answer above clarify things for you? or are you still having problems?

prasad7676, 2017-03-27:
yes. i got it now...but i having issue to pair with hue bridge emulator. i tried many apps to pair but i can only see those accessories in those apps but cant pair with them.

stefanbode, 2017-04-05:
As far as I know, the emulation works as a bridge. Therefore no pairing needed/possible. This makes is easy to integrate with ALEXA. Otherwise, you need a bridge to get it working. I have seen on another project, that they defined the sonoff as a regular Hue client and you need e.g. a running OpenHAB Hue bridge emulator to connect first and then can search with Alexa. Besides the limit of 16 bridges in a network, the current implementation is much more convenient

davidelang, 2017-04-23:
the sonoff doesn't implement full hue emulation, it implements just enough for alexa and google home to find it.

closed 2017-04-23 by davidelang

nunoronconcouto, 2017-10-10:
Hi!!

My google Assistant dont find my sonoff with hue emulation...
Anyone can help me??

Thanks