arendst/Tasmota #282: Lock Configuration - Production Mode

ageorgios, 2017-03-25:
Thanx for the great work!

There would be great if there was a switch on the main page and an API call that would lock the configuration for production mode and do not accept any configuration change API.

This has these benefits:
- Avoid accidental configuration change on currently working Tasmotas
- If the lock API accepts password, there could be roles of admin/user

Thank you.

davidelang, 2017-03-25:
you can put a password on the web console, and if you have a rogue person 
generating mqtt messages, I think you have bigger problems than worrying about 
them changing configs (and the mqtt server can require a password as well)

for 'production' use, you can just disable the webserver entirely, which will 
save a lot of flash and ram.

how would an additional password help?

davidelang, 2017-03-27:
please reopen if you have additional information that would explain better why this complexity would help over the existing options.

closed 2017-03-27 by davidelang