arendst/Tasmota #283: Blinds implementation

ageorgios, 2017-03-25:
Hello, 

I would like to implement a blinds system using a sonoff th16 with 2 relays, 1=moveup 2=movedown.
I can use the PulseTime to have each relay start and stop automatically after X seconds that the blinds take to fully close/open.

1) But blinds need a way to get this information:
- getPercentagePosititon() 
where I need to get the state where the blinds are for example 60% closed.

2) There is also a need to have hardware switches that change also the state

Has anybody implemented something similar?

Is there a roadmap to implement a variable that saves the percentage of opening/closing blinds?

Thank you for your great work.

davidelang, 2017-03-25:
There is not currently something like this. you may be able to rig something up 
with switch inputs hooked to a rotery encoder.

The trouble is that you really can't count on a given amount of motor running 
time moving the blinds a given amount, you need to measure it.

There have been people talking about this, but nothing posted about their 
results.

I think it would be useful to define encoder1a/encoder1b/encoder1stop as input 
pins and then a way to read the encoder position.

goncalvesdiogo, 2017-03-27:
I would like to suggest an idea. If you are using blind rollers, you can reset the starting position using infrared proximity sensors. Knowing the zero point(blinds openned) you can count how many turns the servo motor is needed to close your blind. So your comands are always the "RESET"(open) and the "CLOSE" maybe an STOP one so you can partially close or open the blinds. In case of a module reset due a energy loss.... always starts your programm with a RESET comand so you have the positions reseted.

davidelang, 2017-03-27:
see #288 for a full writeup of what I think is desired

davidelang, 2017-04-02:
closing this in favor of the more general outline in #288

closed 2017-04-02 by davidelang