arendst/Tasmota #284: Electrodragon woes

CommodoreWhite, 2017-03-25:
I have several issues with my v1.5 version of the electrodragon dual switch running 4.0.5. of this
splendid software which i compiled and uploaded using programming.io. I'm hoping if i list them someone might identify what I've done wrong.

1. I've successfully updated a sonoff basic using smadds web site. If i go through the same process on my electrodragon, nothing happens

2. The NTP service doesn't work, timestamps are based on the Unix epoch

3. Poweronstate doesn't work. I've switched off my mqtt service to ensure it doesn't​ get involved. I set poweronstate to ON and cycle the device but it stubbornly stays off, my sonoff basic works fine. The command shows the parameter as stored correctly.

Help would be appreciated.

davidelang, 2017-03-25:
for the ntp issue, check what ntpserver1, ntpserver2, ntpserver3 respond. There 
was a bug around that version where ntp servers became a config option, but it 
wasn't stored in the flash (try with the latest version and see if you still 
have a problem)
k

CommodoreWhite, 2017-03-25:
Thanks. I managed to update the device using file upload rather than web. Now running version 4.1.0 so that's good. Pleased to report that NTP now works so log entries are correctly timestamped.

Poweronstate still doesn't work. I'm going to need that functionality to ensure my house fails-safe in the event WiFi, mqtt, or node-red loses the plot.  Such fails tend to have a poor WAF in my experience

davidelang, 2017-03-27:
please provide the output of 

poweronstate
status0
and the logs during startup (ideally set to 'more debugging') in another issue, someone found that their mqtt/home automation software was overrriding the sonoff's startup state.

CommodoreWhite, 2017-03-28:
Here is the startup and status0 log. Part my fault, I've only just found the savestate command. I've now found poweronstate is honoured for channel 2 but not channel 1. Is this intended?

I've turned mqtt off to ensure isolation from domoticz and node-red.

Peter

00:00:00 APP: Project sonoff Electro-105 (Topic Electro-105, Fallback DVES_075369, GroupTopic electros) Version 4.1.0
00:00:00 Wifi: Connecting to AP1 Police-Surveillance-CAM-1 in mode 11N as Electro-105-4969...
00:00:03 Wifi: Connected
00:00:03 mDNS: Initialized
00:00:03 HTTP: Webserver active on Electro-105-4969.local with IP address 192.168.1.105
00:00:03 RSLT: INFO1 = {"Module":"ElectroDragon", "Version":"4.1.0", "FallbackTopic":"DVES_075369", "GroupTopic":"electros"}
00:00:03 RSLT: INFO2 = {"WebserverMode":"Admin", "Hostname":"Electro-105-4969", "IPaddress":"192.168.1.105"}
00:00:03 RSLT: INFO3 = {"Started":"External System"}
09:31:31 RSLT: RESULT = {"POWER1":"OFF"}
09:31:31 RSLT: POWER1 = OFF
09:31:31 RSLT: RESULT = {"POWER2":"ON"}
09:31:31 RSLT: POWER2 = ON
09:31:38 RSLT: STATE = {"Time":"2017-03-28T09:31:38", "Uptime":0, "Vcc":3.207, "POWER1":"OFF", "POWER2":"ON", "Wifi":{"AP":1, "SSID":"Police-Surveillance-CAM-1", "RSSI":70}}
09:34:32 CMND: Status 0
09:34:32 RSLT: STATUS = {"Status":{"Module":15, "FriendlyName":"Electro-105", "Topic":"Electro-105", "ButtonTopic":"0", "Subtopic":"POWER", "Power":2, "PowerOnState":1, "LedState":1, "SaveData":1, "SaveState":0, "ButtonRetain":0, "PowerRetain":0}}
09:34:32 RSLT: STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"electros", "OtaUrl":"http://s", "Uptime":0, "Sleep":0, "BootCount":57, "SaveCount":165}}
09:34:32 RSLT: STATUS2 = {"StatusFWR":{"Program":"4.1.0", "BuildDateTime":"2017-03-25T16:28:19", "Boot":0, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
09:34:32 RSLT: STATUS3 = {"StatusLOG":{"Seriallog":2, "Weblog":2, "Syslog":0, "LogHost":"192.168.1.2", "SSId1":"Police-Surveillance-CAM-1", "SSId2":"Police-Surveillance-CAM-0", "TelePeriod":300}}
09:34:32 RSLT: STATUS4 = {"StatusMEM":{"ProgramSize":465, "Free":472, "Heap":27, "SpiffsStart":940, "SpiffsSize":64, "FlashSize":4096, "ProgramFlashSize":1024, "FlashChipMode":0}}
09:34:32 RSLT: STATUS5 = {"StatusNET":{"Host":"Electro-105-4969", "IP":"192.168.1.105", "Gateway":"192.168.1.254", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.254", "Mac":"A0:20:A6:07:53:69", "Webserver":2, "WifiConfig":3}}
09:34:32 RSLT: STATUS7 = {"StatusTIM":{"UTC":"Tue Mar 28 08:34:32 2017", "Local":"Tue Mar 28 09:34:32 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
09:34:32 RSLT: STATUS10 = {"StatusSNS":{"Time":"2017-03-28T09:34:32"}}
09:34:32 RSLT: STATUS11 = {"StatusPWR":{"Time":"2017-03-28T09:34:32", "Uptime":0, "Vcc":3.214, "POWER1":"OFF", "POWER2":"ON", "Wifi":{"AP":1, "SSID":"Police-Surveillance-CAM-1", "RSSI":54}}}

arendst, 2017-03-29:
Are you able to control the relays with power commands?
Does Power1 on really turn relay 1 on?
Does Power2 on really turn relay 2 on?

Do you have a schematic of the ElectroDragon v1.5?

Just tested PowerOnState with a Sonoff 4CH and it works fine: All relays are powered on.

CommodoreWhite, 2017-03-29:
Curiouser, and curiouser. 

Toggle 1 switches relay 1, and 2 switches 2. And no, i don't have a diagram.

What's odd is, when i power up the dragon, neither of the relays operate and the status 0 LED is illuminated. The web interface shows relay 2 as on as indeed does the log which you have evidenced. Subsequently, pressing the toggle 1 and 2 buttons on the web interface switches the appropriate relay.

Time to break out another module methinks. I'll get back to you.

Thanks for sticking with me on this.

Btw. I'm confused how poweronstate and savestate interact. Which takes precedence?

CommodoreWhite, 2017-03-29:
Different dragon, similar results, poweronstate is not honoured.

Here's the log.

17:48:35 MQTT: Connected
17:48:35 MQTT: tele/electro-01/LWT = Online (retained)
17:48:35 MQTT: cmnd/electro-01/POWER = 
17:48:35 MQTT: tele/electro-01/INFO1 = {"Module":"ElectroDragon", "Version":"4.1.0", "FallbackTopic":"DVES_075EDB", "GroupTopic":"sonoffs"}
17:48:35 MQTT: tele/electro-01/INFO2 = {"WebserverMode":"Admin", "Hostname":"electro-01-7899", "IPaddress":"192.168.1.101"}
17:48:35 MQTT: tele/electro-01/INFO3 = {"Started":"External System"}
17:48:36 MQTT: stat/electro-01/RESULT = {"POWER1":"OFF"}
17:48:36 MQTT: stat/electro-01/POWER1 = OFF
17:48:36 MQTT: stat/electro-01/RESULT = {"POWER2":"OFF"}
17:48:36 MQTT: stat/electro-01/POWER2 = OFF
17:48:43 MQTT: tele/electro-01/STATE = {"Time":"2017-03-29T17:48:43", "Uptime":0, "Vcc":3.061, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"Police-Surveillance-CAM-1", "RSSI":62}}
17:50:09 HTTP: Logging Seriallog 2, Weblog 4, Syslog 0, Host 192.168.1.2, Port 514, TelePeriod 300
17:50:09 HTTP: Handle config
17:50:10 Config: Saved configuration (1364 bytes) to flash at F7 and count 152
17:50:12 Wifi: Checking connection...
17:50:12 Wifi: Connected
17:50:27 HTTP: Handle root
17:50:32 Wifi: Checking connection...
17:50:32 Wifi: Connected
17:50:34 HTTP: Handle console
17:50:45 CMND: Poweronstate
17:50:45 RSLT: Receive topic cmnd/electro-01/Poweronstate, data size 0, data 
17:50:45 RSLT: DataCb Topic electro-01, Group 0, Index 1, Type POWERONSTATE, Data  ()
17:50:45 MQTT: stat/electro-01/RESULT = {"PowerOnState":1}
17:50:52 Wifi: Checking connection...
17:50:52 Wifi: Connected
17:51:12 Wifi: Checking connection...
17:51:12 Wifi: Connected
17:51:30 CMND: Savestate
17:51:30 RSLT: Receive topic cmnd/electro-01/Savestate, data size 0, data 
17:51:30 RSLT: DataCb Topic electro-01, Group 0, Index 1, Type SAVESTATE, Data  ()
17:51:30 MQTT: stat/electro-01/RESULT = {"SaveState":"OFF"}
17:51:32 Wifi: Checking connection...
17:51:32 Wifi: Connected
17:51:52 Wifi: Checking connection...
17:51:52 Wifi: Connected
17:52:12 Wifi: Checking connection...
17:52:12 Wifi: Connected
17:52:32 Wifi: Checking connection...
17:52:32 Wifi: Connected

arendst, 2017-03-29:
savestate off means no power state is saved. A poweronstate of toggle will therefoe never work.

Suggest to leave powersave set to 1 and try again with poweronstate set to 1. You will need to remove power and reapply power for poweronstate to work; a restart only will not work.

CommodoreWhite, 2017-03-29:
Sorry, forgot to include the status 0 output last time.

00:00:00 Config: Loaded configuration from flash at F7 and count 152
00:00:00 APP: Bootcount 31
00:00:00 APP: Project sonoff Electro-01 (Topic electro-01, Fallback DVES_075EDB, GroupTopic sonoffs) Version 4.1.0
00:00:00 Config: Saved configuration (1364 bytes) to flash at F8 and count 153
00:00:00 Wifi: Checking connection...
00:00:00 Wifi: Attempting connection...
00:00:00 Wifi: Patch issue 2186
00:00:00 Wifi: Connecting to AP1 Police-Surveillance-CAM-1 in mode 11N as electro-01-7899...
00:00:01 Wifi: Checking connection...
00:00:01 Wifi: Attempting connection...
00:00:02 Wifi: Checking connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Checking connection...
00:00:03 Wifi: Connected
00:00:03 mDNS: Initialized
00:00:03 HTTP: Webserver active on electro-01-7899.local with IP address 192.168.1.101
00:00:03 RTC: (UTC) Wed Mar 29 16:54:55 2017
00:00:03 RTC: (DST) Sun Mar 26 02:00:00 2017
00:00:03 RTC: (STD) Sun Oct 29 03:00:00 2017
17:54:57 MQTT: Attempting connection...
17:54:59 mDNS: Query done with 0 mqtt services found
17:54:59 MQTT: Connected
17:54:59 MQTT: tele/electro-01/LWT = Online (retained)
17:54:59 MQTT: cmnd/electro-01/POWER = 
17:54:59 MQTT: tele/electro-01/INFO1 = {"Module":"ElectroDragon", "Version":"4.1.0", "FallbackTopic":"DVES_075EDB", "GroupTopic":"sonoffs"}
17:54:59 MQTT: tele/electro-01/INFO2 = {"WebserverMode":"Admin", "Hostname":"electro-01-7899", "IPaddress":"192.168.1.101"}
17:54:59 MQTT: tele/electro-01/INFO3 = {"Started":"External System"}
18:49:38 Wifi: Checking connection...
18:49:38 Wifi: Connected
18:49:58 Wifi: Checking connection...
18:49:58 Wifi: Connected
18:50:18 Wifi: Checking connection...
18:50:18 Wifi: Connected
18:50:38 Wifi: Checking connection...
18:50:38 Wifi: Connected
18:50:58 Wifi: Checking connection...
18:50:58 Wifi: Connected
18:51:18 Wifi: Checking connection...
18:51:18 Wifi: Connected
18:51:21 CMND: Status 0
18:51:21 RSLT: Receive topic cmnd/dragon-101/Status, data size 1, data 0
18:51:21 RSLT: DataCb Topic dragon-101, Group 0, Index 1, Type STATUS, Data 0 (0)
18:51:21 RSLT: STATUS = {"Status":{"Module":15, "FriendlyName":"Dragon-101", "Topic":"dragon-101", "ButtonTopic":"0", "Subtopic":"POWER", "Power":0, "PowerOnState":1, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
18:51:21 RSLT: STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"dragons", "OtaUrl":"http://192.168.1.2:80/api/arduino/sonoff.ino.bin", "Uptime":0, "Sleep":0, "BootCount":37, "SaveCount":165}}
18:51:21 RSLT: STATUS2 = {"StatusFWR":{"Program":"4.1.0", "BuildDateTime":"2017-03-25T16:28:19", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
18:51:21 RSLT: STATUS3 = {"StatusLOG":{"Seriallog":2, "Weblog":4, "Syslog":0, "LogHost":"192.168.1.2", "SSId1":"Police-Surveillance-CAM-1", "SSId2":"Police-Surveillance-CAM-0", "TelePeriod":300}}
18:51:21 RSLT: STATUS4 = {"StatusMEM":{"ProgramSize":465, "Free":472, "Heap":25, "SpiffsStart":940, "SpiffsSize":64, "FlashSize":4096, "ProgramFlashSize":1024, "FlashChipMode":0}}
18:51:21 RSLT: STATUS5 = {"StatusNET":{"Host":"dragon-101-7899", "IP":"192.168.1.101", "Gateway":"192.168.1.254", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.254", "Mac":"A0:20:A6:07:5E:DB", "Webserver":2, "WifiConfig":3}}
18:51:21 RSLT: STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 29 17:51:21 2017", "Local":"Wed Mar 29 18:51:21 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
18:51:21 RSLT: STATUS10 = {"StatusSNS":{"Time":"2017-03-29T18:51:21"}}
18:51:21 RSLT: STATUS11 = {"StatusPWR":{"Time":"2017-03-29T18:51:21", "Uptime":0, "Vcc":3.052, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"Police-Surveillance-CAM-1", "RSSI":68}}}
18:51:38 Wifi: Checking connection...
18:51:38 Wifi: Connected
18:51:58 Wifi: Checking connection...
18:51:58 Wifi: Connected
18:52:18 Wifi: Checking connection...
18:52:18 Wifi: Connected

CommodoreWhite, 2017-03-29:
FYI A third dragon seems to exhibit the same behaviour.

00:00:02 Wifi: Checking connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Checking connection...
00:00:03 Wifi: Connected
00:00:03 mDNS: Initialized
00:00:03 HTTP: Webserver active on electro-02-3006.local with IP address 192.168.1.102
00:00:03 RSLT: INFO1 = {"Module":"ElectroDragon", "Version":"4.1.0", "FallbackTopic":"DVES_07ABBE", "GroupTopic":"sonoffs"}
00:00:03 RSLT: INFO2 = {"WebserverMode":"Admin", "Hostname":"electro-02-3006", "IPaddress":"192.168.1.102"}
00:00:03 RSLT: INFO3 = {"Started":"External System"}
00:00:03 RTC: (UTC) Wed Mar 29 20:32:48 2017
00:00:03 RTC: (DST) Sun Mar 26 02:00:00 2017
00:00:03 RTC: (STD) Sun Oct 29 03:00:00 2017
20:32:50 RSLT: RESULT = {"POWER1":"OFF"}
20:32:50 RSLT: POWER1 = OFF
20:32:50 RSLT: RESULT = {"POWER2":"OFF"}
20:32:50 RSLT: POWER2 = OFF
20:32:57 RSLT: STATE = {"Time":"2017-03-29T20:32:57", "Uptime":0, "Vcc":3.177, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"Police-Surveillance-CAM-1", "RSSI":62}}
20:33:08 Wifi: Checking connection...
20:33:08 Wifi: Connected
20:33:12 HTTP: Handle console
20:33:20 CMND: Status 0
20:33:20 RSLT: Receive topic cmnd/electro-02/Status, data size 1, data 0
20:33:20 RSLT: DataCb Topic electro-02, Group 0, Index 1, Type STATUS, Data 0 (0)
20:33:20 RSLT: STATUS = {"Status":{"Module":15, "FriendlyName":"Electro-02", "Topic":"electro-02", "ButtonTopic":"0", "Subtopic":"POWER", "Power":0, "PowerOnState":1, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
20:33:20 RSLT: STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://192.168.1.2:80/api/arduino/sonoff.ino.bin", "Uptime":0, "Sleep":0, "BootCount":26, "SaveCount":60}}
20:33:20 RSLT: STATUS2 = {"StatusFWR":{"Program":"4.1.0", "BuildDateTime":"2017-03-25T16:28:19", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
20:33:20 RSLT: STATUS3 = {"StatusLOG":{"Seriallog":2, "Weblog":4, "Syslog":0, "LogHost":"domus1", "SSId1":"Police-Surveillance-CAM-1", "SSId2":"Police-Surveillance-CAM-0", "TelePeriod":300}}
20:33:20 RSLT: STATUS4 = {"StatusMEM":{"ProgramSize":465, "Free":472, "Heap":27, "SpiffsStart":940, "SpiffsSize":64, "FlashSize":4096, "ProgramFlashSize":1024, "FlashChipMode":0}}
20:33:20 RSLT: STATUS5 = {"StatusNET":{"Host":"electro-02-3006", "IP":"192.168.1.102", "Gateway":"192.168.1.254", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.254", "Mac":"A0:20:A6:07:AB:BE", "Webserver":2, "WifiConfig":3}}
20:33:20 RSLT: STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 29 20:33:20 2017", "Local":"Wed Mar 29 20:33:20 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":0}}
20:33:20 RSLT: STATUS10 = {"StatusSNS":{"Time":"2017-03-29T20:33:20"}}
20:33:20 RSLT: STATUS11 = {"StatusPWR":{"Time":"2017-03-29T20:33:20", "Uptime":0, "Vcc":3.165, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"Police-Surveillance-CAM-1", "RSSI":74}}}
20:33:29 Wifi: Checking connection...
20:33:29 Wifi: Connected
20:33:49 Wifi: Checking connection...
20:33:49 Wifi: Connected
20:34:09 Wifi: Checking connection...
20:34:09 Wifi: Connected

CommodoreWhite, 2017-03-29:
And another

00:00:03 mDNS: Initialized
00:00:03 HTTP: Webserver active on dragon-103-0088.local with IP address 192.168.1.103
00:00:03 RTC: (UTC) Wed Mar 29 22:02:56 2017
00:00:03 RTC: (DST) Sun Mar 26 02:00:00 2017
00:00:03 RTC: (STD) Sun Oct 29 03:00:00 2017
22:02:58 MQTT: Attempting connection...
22:03:00 mDNS: Query done with 0 mqtt services found
22:03:00 MQTT: Connected
22:03:00 MQTT: tele/dragon-103/LWT = Online (retained)
22:03:00 MQTT: cmnd/dragon-103/POWER = 
22:03:00 MQTT: tele/dragon-103/INFO1 = {"Module":"ElectroDragon", "Version":"4.1.0", "FallbackTopic":"DVES_148058", "GroupTopic":"dragons"}
22:03:00 MQTT: tele/dragon-103/INFO2 = {"WebserverMode":"Admin", "Hostname":"dragon-103-0088", "IPaddress":"192.168.1.103"}
22:03:00 MQTT: tele/dragon-103/INFO3 = {"Started":"External System"}
22:03:01 MQTT: stat/dragon-103/RESULT = {"POWER1":"OFF"}
22:03:01 MQTT: stat/dragon-103/POWER1 = OFF
22:03:01 MQTT: stat/dragon-103/RESULT = {"POWER2":"OFF"}
22:03:01 MQTT: stat/dragon-103/POWER2 = OFF
22:03:06 HTTP: Handle root
22:03:08 MQTT: tele/dragon-103/STATE = {"Time":"2017-03-29T22:03:08", "Uptime":0, "Vcc":3.111, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"Police-Surveillance-CAM-1", "RSSI":48}}
22:03:10 HTTP: Handle console
22:03:17 Wifi: Checking connection...
22:03:17 Wifi: Connected
22:03:19 CMND: Status 0
22:03:19 RSLT: Receive topic cmnd/dragon-103/Status, data size 1, data 0
22:03:19 RSLT: DataCb Topic dragon-103, Group 0, Index 1, Type STATUS, Data 0 (0)
22:03:19 MQTT: stat/dragon-103/STATUS = {"Status":{"Module":15, "FriendlyName":"Dragon-103", "Topic":"dragon-103", "ButtonTopic":"0", "Subtopic":"POWER", "Power":0, "PowerOnState":1, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
22:03:19 MQTT: stat/dragon-103/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"dragons", "OtaUrl":"http://192.168.1.2:80/api/arduino/sonoff.ino.bin", "Uptime":0, "Sleep":0, "BootCount":13, "SaveCount":34}}
22:03:19 MQTT: stat/dragon-103/STATUS2 = {"StatusFWR":{"Program":"4.1.0", "BuildDateTime":"2017-03-25T16:28:19", "Boot":0, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
22:03:19 MQTT: stat/dragon-103/STATUS3 = {"StatusLOG":{"Seriallog":2, "Weblog":4, "Syslog":0, "LogHost":"domus1", "SSId1":"Police-Surveillance-CAM-1", "SSId2":"Police-Surveillance-CAM-0", "TelePeriod":300}}
22:03:19 MQTT: stat/dragon-103/STATUS4 = {"StatusMEM":{"ProgramSize":465, "Free":472, "Heap":26, "SpiffsStart":940, "SpiffsSize":64, "FlashSize":4096, "ProgramFlashSize":1024, "FlashChipMode":0}}
22:03:19 MQTT: stat/dragon-103/STATUS5 = {"StatusNET":{"Host":"dragon-103-0088", "IP":"192.168.1.103", "Gateway":"192.168.1.254", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.254", "Mac":"A0:20:A6:14:80:58", "Webserver":2, "WifiConfig":3}}
22:03:19 MQTT: stat/dragon-103/STATUS6 = {"StatusMQT":{"Host":"192.168.1.2", "Port":1883, "ClientMask":"DVES_%06X", "Client":"DVES_148058", "User":"admin", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
22:03:19 MQTT: stat/dragon-103/STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 29 22:03:19 2017", "Local":"Wed Mar 29 22:03:19 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":0}}
22:03:19 MQTT: stat/dragon-103/STATUS10 = {"StatusSNS":{"Time":"2017-03-29T22:03:19"}}
22:03:19 MQTT: stat/dragon-103/STATUS11 = {"StatusPWR":{"Time":"2017-03-29T22:03:19", "Uptime":0, "Vcc":3.112, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"Police-Surveillance-CAM-1", "RSSI":62}}}
22:03:38 Wifi: Checking connection...
22:03:38 Wifi: Connected
22:03:48 CMND: Poweronstate
22:03:48 RSLT: Receive topic cmnd/dragon-103/Poweronstate, data size 0, data 
22:03:48 RSLT: DataCb Topic dragon-103, Group 0, Index 1, Type POWERONSTATE, Data  ()
22:03:48 MQTT: stat/dragon-103/RESULT = {"PowerOnState":1}

arendst, 2017-03-30:
And now with  a power cycle instead of a restart as I noticed e few answers ago...

CommodoreWhite, 2017-03-30:
Btw i only ever power cycle the devices, never restart when doing these tests for you.

Peter

arendst, 2017-03-30:
Well as long as INFO3 returns the wrong message it will never work in your case.

Power off. Wait a minute and power on again.

CommodoreWhite, 2017-03-30:
Aha! Might not have been giving the units enough time to power reset. That said, the unit seems to go through the reset process. I'll try again when i get home.

CommodoreWhite, 2017-03-30:
Here is the log having switched off the unit for a few minutes. What message should i be expecting info 3?

00:00:00 Wifi: Connecting to AP1 Police-Surveillance-CAM-1 in mode 11N as dragon-103-0088...
00:00:01 Wifi: Checking connection...
00:00:01 Wifi: Attempting connection...
00:00:02 Wifi: Checking connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Checking connection...
00:00:03 Wifi: Connected
00:00:03 mDNS: Initialized
00:00:03 HTTP: Webserver active on dragon-103-0088.local with IP address 192.168.1.103
00:00:03 RSLT: INFO1 = {"Module":"ElectroDragon", "Version":"4.1.1", "FallbackTopic":"DVES_148058", "GroupTopic":"dragons"}
00:00:03 RSLT: INFO2 = {"WebserverMode":"Admin", "Hostname":"dragon-103-0088", "IPaddress":"192.168.1.103"}
00:00:03 RSLT: INFO3 = {"Started":"External System"}
00:00:03 RTC: (UTC) Thu Mar 30 16:46:23 2017
00:00:03 RTC: (DST) Sun Mar 26 02:00:00 2017
00:00:03 RTC: (STD) Sun Oct 29 03:00:00 2017
17:46:25 RSLT: RESULT = {"POWER1":"OFF"}
17:46:25 RSLT: POWER1 = OFF
17:46:25 RSLT: RESULT = {"POWER2":"OFF"}
17:46:25 RSLT: POWER2 = OFF
17:46:34 HTTP: Handle console
17:46:35 RSLT: STATE = {"Time":"2017-03-30T17:46:35", "Uptime":0, "Vcc":3.116, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"Police-Surveillance-CAM-1", "RSSI":56}}
17:46:47 Wifi: Checking connection...
17:46:47 Wifi: Connected
17:46:58 CMND: Status 0
17:46:58 RSLT: Receive topic cmnd/dragon-103/Status, data size 1, data 0
17:46:58 RSLT: DataCb Topic dragon-103, Group 0, Index 1, Type STATUS, Data 0 (0)
17:46:58 RSLT: STATUS = {"Status":{"Module":15, "FriendlyName":"Dragon-103", "Topic":"dragon-103", "ButtonTopic":"0", "Power":0, "PowerOnState":1, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
17:46:58 RSLT: STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"dragons", "OtaUrl":"http://192.168.1.2:80/api/arduino/sonoff.ino.bin", "Uptime":0, "Sleep":0, "BootCount":21, "SaveCount":75}}
17:46:58 RSLT: STATUS2 = {"StatusFWR":{"Program":"4.1.1", "BuildDateTime":"2017-03-29T16:45:03", "Boot":0, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
17:46:58 RSLT: STATUS3 = {"StatusLOG":{"Seriallog":2, "Weblog":4, "Syslog":0, "LogHost":"domus1", "SSId1":"Police-Surveillance-CAM-1", "SSId2":"Police-Surveillance-CAM-0", "TelePeriod":300}}
17:46:58 RSLT: STATUS4 = {"StatusMEM":{"ProgramSize":464, "Free":472, "Heap":27, "SpiffsStart":940, "SpiffsSize":64, "FlashSize":4096, "ProgramFlashSize":1024, "FlashChipMode":0}}
17:46:58 RSLT: STATUS5 = {"StatusNET":{"Host":"dragon-103-0088", "IP":"192.168.1.103", "Gateway":"192.168.1.254", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.254", "Mac":"A0:20:A6:14:80:58", "Webserver":2, "WifiConfig":3}}
17:46:58 RSLT: STATUS7 = {"StatusTIM":{"UTC":"Thu Mar 30 16:46:58 2017", "Local":"Thu Mar 30 17:46:58 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
17:46:58 RSLT: STATUS10 = {"StatusSNS":{"Time":"2017-03-30T17:46:58"}}
17:46:58 RSLT: STATUS11 = {"StatusSTS":{"Time":"2017-03-30T17:46:58", "Uptime":0, "Vcc":3.117, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"Police-Surveillance-CAM-1", "RSSI":52}}}
17:47:09 Wifi: Checking connection...
17:47:09 Wifi: Connected
17:47:13 CMND: Poweronstate
17:47:13 RSLT: Receive topic cmnd/dragon-103/Poweronstate, data size 0, data 
17:47:13 RSLT: DataCb Topic dragon-103, Group 0, Index 1, Type POWERONSTATE, Data  ()
17:47:13 RSLT: RESULT = {"PowerOnState":1}
17:47:29 Wifi: Checking connection...
17:47:29 Wifi: Connected
17:47:49 Wifi: Checking connection...
17:47:49 Wifi: Connected

arendst, 2017-03-30:
See my previous answer info3. Weird it still responds with external system...

CommodoreWhite, 2017-03-30:
Sorry, misread! Don't think I've ever seen a power on type message when using electrodragons. I'll try with my sonoff just to see what i should be expecting. Sorry this is dragging on.

CommodoreWhite, 2017-03-30:
Ok. Just tried with dragon that i had loaded your software onto a week ago. Looked at the console l log and found that it reported "started external system". Seems to me that version 1.5 of the dragons don't correctly report how they've been powered up. Perhaps someone with earlier versions of these​ devices can report their experiences.

All 5 of my rev 1.5 dragons seem to behave the same way

Thanks again for your patience.

CommodoreWhite, 2017-04-01:
Can you tell me the criteria the dragons use to determine how they've been started up. I don't understand what "started external system" means anyway, please enlighten!

CommodoreWhite, 2017-04-01:
What does"started external system" mean? How do you differentiate between the various startup modes?

Regards

arendst, 2017-04-01:
I rely on the Arduino-ESP8266 library which in turn relies on the Espressif provided SDK. The SDK should respond with the correct reboot status but for some reason fails to detect a power cycle on your ElectroDragon. There were issues with previous SDK's as can be found on the issues page of the Arduino-ESP8266 library but they should have been solved for the current used SDK 1.5.3.

CommodoreWhite, 2017-04-01:
Presumably I'm neither the first nor the last to encounter this issue so the SDK will get fixed in the fullness of time. Thanks for your patience.

Peter

CommodoreWhite, 2017-04-15:
Hi, a couple of things:
1 I've found the code that implements poweronstate. What's the smallest change you'd recommend to force my dragons to always power-up in an On state,
2  I'm not sure the issue is with the SDK because my sonoffs work a treat. Is it more likely that the dragons do not respond correctly, are you aware of other dragon users that have a different experience to me?

Thanks again, Peter

davidelang, 2017-04-15:
On Sat, 15 Apr 2017, CommodoreWhite wrote:

> 1 I've found the code that implements poweronstate. What's the smallest change you'd recommend to force my dragons to always power-up in an On state,

you should not need to make any code changes, you should be able to just set the 
poweronstate in the config.

CommodoreWhite, 2017-04-16:
... except that the poweronstate setting depends on the cpu returning "power on" as the reset reason. for some reason my dragons return "external system" so poweronstate is never actioned. c.f.

`  if (ESP.getResetReason() == "Power on") {
`

Regards

CommodoreWhite, 2017-04-16:
Right! I've successfully used platform.io in Windows to upload TASMOTA to my electrodragon switch. I'd hoped it was both the first and last time because a) I found the programming environment confusing, and b) you provide firmware.bin files which I've since used to upgrade my dragons using the web interface.

I'd like to create my own firmware.bin file, but can't figure the magic commands. A clue would be appreciated.

davidelang, 2017-04-23:
firmware.bin is created by the platformio build process. Using the arduino IDE it build the fle sonoff.ino.bin instead. go through the "getting started" pages in order.

I'm closing this issue as it's now not electrodragon related. open another issue if you continue to have build problems (and if the docs are confusing, please help us make them better once you figure out how things work)

closed 2017-04-23 by davidelang

CommodoreWhite, 2017-04-24:
it sort of is electrodragon related because i wouldn't have had to do any
of this if my dragons produced the correct message at startup as you have
seen in the logs. Nevertheless, happy to close the issue since the problem
doesn't seem to be with your code.

Peter

CommodoreWhite, 2017-04-24:
Yep. The time problem vanished on later versions of tasmota. Thanks.

arendst, 2017-04-28:
Will test for "External System" too when powered on in next version