arendst/Tasmota #285: I found there is some cool effect for WS2812, can it be implement into Tasmota?

hamwong, 2017-03-25:
I found a code with some cool effect for WS2812, can it be implement into Tasmota?

https://github.com/bruhautomation/ESP-MQTT-Digital-LEDs

also I am thinking, will it possibile to add a page in webui than user may customize few effects, just upload the code to the page and assign command name, like effect 1?

davidelang, 2017-03-27:
similar to #39 It would depend  on what's needed for these effects. The ws2812 modules need to have the data fed to them on an ongoing basis, and there isn't time to calculate it on the fly. So the complete sequence needs to be stored on the device. Since we are so tight on ram and flash, this could be a problem.

davidelang, 2017-03-27:
how small is "real small", even a few Kb can be a dealbreaker

remember, we have no ability to change the hardware in these devices, so we 
can't upgrade the flash, that would take using a different device. There are a 
few devices that have more flash, but the default is to build for the small 
flash size (and we actually use less than half of that, so that we can do an OTA 
update)

I'm pretty sure that arendst would accept a PR that enabled this, as long as it 
was ifdef off by default and documented the size difference that it added. It 
could possibly require an ifdef test to make sure it was being built on a device 
with enough flash.

David Lang

hamwong, 2017-03-27:
Sure, it is up to you and @arendst, I just feel ws2812 is a good add on but function is very limited. It will be nice to have a little bit more feature on it

davidelang, 2017-03-27:
it's not up to me :-) I'm just assisting.

It's up to whoever writes/ports the code and arendst to decide if he will accept 
the code.

I'm just trying to help identify what would probably be needed to have it added.

hamwong, 2017-03-27:
Ok. So I guess this would shounds like no. I'll close the case than

davidelang, 2017-03-27:
It's not a no, it's a "I think it's unlikely that @arendst will do this" and an 
explination of how I think it could be done.

On Mon, 27 Mar 2017, hamwong wrote:

> Date: Mon, 27 Mar 2017 12:11:14 -0700
> From: hamwong <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f66516568698ad6fb324229f065892243699617af92cf0000000114f125d
>     292a169ce0cef119c@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] I found there is some cool effect for
>     WS2812, can it be implement into Tasmota? (#285)
> 
> Ok. So I guess this would shounds like no. I'll close the case than
>
>

davidelang, 2017-03-27:
This is a reasonably common request, so unless @arendst says that he wouldn't accept it, I'm going to leave it open as a reference for others asking similar questions

ascillato2, 2018-07-10:
Duplicated issue. Closing.

closed 2018-07-10 by ascillato2