arendst/Tasmota #286: Choose for power/state to be published as 0/1 instead of OFF/ON

McCloud, 2017-03-25:
As said above, it would be ideal if I could switch between them or be able to change them.  Some of the MQTT related stuff I'm using is sometimes looking for a 0/1 instead of OFF/ON.

davidelang, 2017-03-25:
This can be set in user_config*

McCloud, 2017-03-25:
Yeah, I already saw that.  Any intent to add that to the commands list or UI so it can be changed without unique compiles?

davidelang, 2017-03-25:
not at this time.

On Sat, 25 Mar 2017, McCloud wrote:

> Yeah, I already saw that.  Any intent to add that to the commands list or UI so it can be changed without unique compiles?

arendst, 2017-03-26:
Will do in next version.

McCloud, 2017-03-30:
Other two commands work, but "statetext2 1" seems to never actually update and keeps putting out "ON" as the setting

arendst, 2017-03-30:
And why do you think it won't work....

RTFW and use only 10 characters at most.

McCloud, 2017-03-30:
You're totally right, for whatever reason a reboot fixed it, before even simple test cases like "14:47:33 CMND: statetext2 ontest" were still returning the previous value. 

Thanks

closed 2017-04-02 by arendst