arendst/Tasmota #287: Power State on WiFi Connection loss

HavoK-at, 2017-03-25:
Hey!

Is it possible to add a default state for power on WiFi connection loss? 
I just want to be sure my stuff is turned off even if the sonoff loses connection.
Short question, what is the purpose of BlinkCount and BlinkTime?

Thanks!

davidelang, 2017-03-27:
There is a default state on poweron, but not on wifi/mqtt connection loss. I suspect that you loose connection a lot more frequently than you imagine, but it's worth experimenting with.

sebastianlorenzen, 2017-04-13:
I have "the same issue" My sonoffs are going on and off after network issues. sometimes if just blinks, sometimes to light stays on for hours.

cduvenhorst, 2017-04-16:
Try to disable persistence in the broker ....
Seems to be a persistence problem.

sebastianlorenzen, 2017-04-16:
I'm using mosquito as broker. Afaik it is off by default. Will double check this.
Any other suggestions?

ascillato2, 2018-07-10:
Hi,

This issue has been addressed. Please search in the issues list. Closing this issue. Please reopen if needed. Thanks.

closed 2018-07-10 by ascillato2