arendst/Tasmota #288: Add position support functions (motor direction/move support, limits, encoders)

davidelang, 2017-03-25:
There have been ongoing questions about adding support for things like blinds, so to consolidate the various requests into one discussion, I've created this issue.

This is not a request for @arendst to implement all of this, but rather as a RFC for people to consider and possibly implement some portions of it in a way that can later be expanded to add later features without ending up in dead-ends.

There are three different pieces involved in defining a positioning object

## Motor support
### none
A position encoder that is driven by something else is useful in itself. So it's possible to define a positioning object with no motor.

An example of this would be a weather station where you have a wind direction encoder (type absolute), a rain gauge encoder (single line, always 'rotates' in one direction), possibly a wind speed encoder (single line, sampled at a regular interval)
### continuous fixed speed
either with a DPDT relay to indicate direction and another relay to turn the motor on/off, or using an h-bridge where one input indicates direction and a second indicates speed
### continuous variable speed
same as continuous fixed speed, but a pwm signal is used to indicate speed
### stepper
driving a stepper motor driver, one pin indicates direction, an second one pulses for every step (with a pseudo encoder that tracks how many steps have been taken)
### fixed servo
set a pwm signal (with some specific constraints) on a pin that indicates the position the servo should move to
### continuous servo
set a pwm signal (with some specific constraints) on a pin that indicates the direction and speed the servo should run at. This requires defining the neutral/center/off position for the servo

## limit switches
### physical
a GPIO to a switch to indicate absolute position
### software
a limit of how far you can go (time or steps)
### none
the motor/sensor can rotate forever. There is a wrap-around value depending on the internal implementation of either 64K (16 bit value) or 4G (32 bit value)
### reset on report
every time the encoder value is reported in a tele/ report, it is reset

# encoders
It is assumed that any needed debouncing is done in hardware so that software doesn't need to worry about it.
### open loop
aka "dead reckoning"

no actual encoder signal, the system tracks either the number of steps (for a stepper motor), or the time (or time * speed) for a continuous motor and reports a 'dead reckoning' calculation to determin the current position
### single line
indicates movement but not direction. position is tracked by assuming that all movement happens in the direction last specified for movement (with a startup direction defined)
### double line
This can detect direction as well as 
double line encoder, indicates direction
### analog
reads the ADC as a position
### absolute
Probably not in the first versions
multiple lines needed, uses grey code to detect position.

grey code is a way of arranging binary sensors so that there is only one line that changes at each position change. This way you don't get errors from sensors not being exactly aligned.
https://en.wikipedia.org/wiki/Gray_code

binary reflected grey code is the default to be used, single track grey code support would be a nice-to-have but probably a different encoder type

# commands
resetEncoder X sets the encoder X position to '0'
readEncoder X returns the current position report from encoder X
Motor<x> direction,distance,speed (or json equivalent) runs the motor, if a limit is hit, the motor stops, if speed is not specified, it runs at full speed. If distance is not specified it runs until it hits the limit

optionally, position information can be saved to flash at the end of a motor action

# configuration
a positioning definition would be send to the device via JSON 

note that this approach of listing gpios as part of the positioning definition is not the standard tasmota approach of a module definition being an array of GPIOs, listing what item is on each one is.

### everything but absolute encoders
1. motor type
1. motor dirction gpio
1. motor speed gpio
1. encoder type
1. encoder line 1 gpio
1. encoder line 2 gpio
1. low limit type 
1. high limit type
1. low encoder gpio
1. high encoder gpio
1. low limit value
1. high limit value
encoder values are not allowed to exceed the limit values, if they do, the limits are reported instead
If a limit is triggerd, the encoder value is set to the limit

A note on the value ranges. 8 bits (0-255) are not sufficiaent, so they need to be at least 16 bit (0-64K), but it may make sense to let them be 24 bits (0-16M). If flash space is not limited, then letting them be 32 bits (0-4G) is reasonable.
With a common stepper motor, you start off with 200 counts/rev and may mutliply it by 8 with microstepping quite reasonably, so that would be 1600 steps/rev. 16 bits only gives you 40 revolutions, while 24 bits would give you > 10k revolutions.

### absolute encoders
1. motor type
1. motor type gpio
1. motor speed gpio
1. encoder type (absolute)
1. number of GPIOs used
1. ordered list of GPIOs

titomontes, 2018-04-30:
Is there any news on this front?
Im VERY interested in helping develop this function, though I´m savvy enough to code.

Anyone?

lobocobra, 2018-07-15:
see 824#

skatun, 2018-07-24:
I do have an arduino code running where I can send number of steps over serial to an esp8266 board connected to a stepper motor shield, I will look into integrating it into tasmota this fall. My plan was to use to control blinds by sending steps to the motor and once in a while (fully open switch) reset it to compensate for the risk of loosing steps or manual override.

ascillato, 2018-08-21:
Just for Reference #3141

svh1985, 2018-11-14:
Would love to see support for motors to control shutters or curtains!

luisdorado, 2019-01-23:
Hi all, I've developed a blind driver for tasmota. It's a controller for rolleshutter motors, not for stepper motors. So the position is stored as elapsed ms from the top. You can configure the time needed for the rollershutter to reach the bottom position, and also an additional time needed to reach the max position (so that the strips are completely joined and no light enters)

https://github.com/luisdorado/Sonoff-Tasmota

Hope you like it and can help me to improve it!

meingraham, 2019-01-23:
Have you considered requesting that this functionality be merged directly into TASMOTA?

luisdorado, 2019-01-23:
I'd like to improve the code first and see if it's useful for more users before requesting a merge to the main project. Any feedback is welcome!

svh1985, 2019-02-05:
@luisdorado I think merging/pull request it into Tasmota will definitely give you a lot of help because of the amount of developers contributing to the project.

meingraham, 2019-02-05:
@luisdorado,

Have you inspected https://github.com/stefanbode/Sonoff-Tasmota? @stefanbode's TASMOTA fork has features for roller shutters. I believe that implementation uses two paired relays to "coordinate" up/down commands. I don't know enough about it to know whether it uses a stepper counter or timing. But, perhaps, other than counting or timing, there may be some similarities whereby your could take advantage of Stefan's work.

Cheers!

Mike

localhost61, 2019-02-20:
@luisdorado 
Did you code something in your repository?

lobocobra, 2019-02-20:
You can have also a look here:
https://github.com/lobocobra/sonoff-shutter

I implemented it as a driver. This might be a solution for your or some help for coding

calcipher49, 2019-02-24:
About shutters, there is also Stefan Bode mode: [](https://github.com/stefanbode/Sonoff-Tasmota)

ridikuel, 2019-03-03:
I have a use case involving a stepper motor driven by an A4988 breakout board. Also I want to use two limit switches to avoid damage and reset in case of messing up with the stepper count.

I forked and though I am not well experienced in coding, I'll give it a try.
https://github.com/ridikuel/Sonoff-Tasmota-stepper

paperreduction, 2019-06-04:
@ridikuel I'm interested in helping get stepper motor support added to Tasmota. I'm not seeing where you modified Tasmota to add A4988 support. https://github.com/ridikuel/Sonoff-Tasmota-stepper/commits?author=ridikuel

I was thinking we could add xdrv_22_stepper.ino, and that would let have generic stepper functionality. We could then hopefully create some device templates for specific breakout boards and stepper drivers.

paperreduction, 2019-06-05:
@arendst I'm open to any feedback you have for me before I start trying to figure out the best way to get stepper motor support added.

ridikuel, 2019-08-27:
@paperreduction Thanks for your interest. I currently working on the idea of multiple MCP230xx support as this popped up for higher priority. But I won't forget on getting back to stepper driver. What might be a good solution, working on my fork? 
Adding xdrv_22_stepper seems good start.

@arendst Do you have recommendations?

shantur, 2019-09-06:
Looking forward to get one of the implementations merged in. I am planning to build one for myself

Jason2866, 2019-09-08:
Stepper driver is done!
https://github.com/arendst/Sonoff-Tasmota/pull/6370

Sergio-tix, 2019-10-02:
How so? Still not supported EC11 / EC12 (KY-040), etc. ??? In a rotary way, it is very convenient to set the brightness of the lighting, temperature. Damn it, turning the knob is much more convenient than standing and poking a button many times. Even roller blinds are more convenient to manage. It is necessary that the encoder, when turning, send a +1 or -1 value to the variable. In order for the server to perform smooth adjustment, we must send each value, even if only 50 values per second.
WiFi will handle such a flow. 
I have TRÅDFRI Wireless dimmer (ZigBee). But it works on a different algorithm. The accelerometer measures the time the washer rotates. He assumes that the puck rotates at the same speed, which means that during the rotation time you can determine how much the puck has turned. And to guess this value is sent. This is a bad algorithm. It does not work well.
So I ask to add support EC11 here.

Sergio-tix, 2019-10-02:
If someone does not know protokol work EC11, There are many examples of Arduino code on the Internet, and other documentation. https://youtu.be/zYE5JhUMjys

meingraham, 2019-10-02:
@Sergio-tix 

With the knowledge you have on the EC11 protocols, and the A4988 stepper motor driver as a template, I propose that you develop the driver for the EC11 and submit a pull request to have it added to Tasmota.

Mike

Sergio-tix, 2019-10-04:
> @ Sergio-Tix
> 
> Обладая знаниями о протоколах EC11 и драйвером шагового двигателя A4988 в качестве шаблона, я предлагаю вам разработать драйвер для EC11 и отправить запрос на извлечение для его добавления в Tasmota.
> 
> Майк

Unfortunately, I am not a programmer. Should I develop a library? And no one else? Can't we use the ready? A lot of them. Here are some:
https://github.com/aleh/AAL-EC11
https://github.com/buxtronix/arduino/tree/master/libraries/Rotary
https://github.com/brianlow/Rotary
https://www.arduinolibraries.info/libraries/encoder
Will there be any problems with the license?

meingraham, 2019-10-05:
@Sergio-tix 

Yes, if you need/want the EC11 adapter to work with Tasmota, you will have to develop it or enlist someone to do it for you.

https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#can-you-add-this-unsupported-sensor-to-tasmota

A great deal of the features in Tasmota were added by a user who wanted/needed the driver for themselves. They donated their time to take the documentation and existing code libraries and developed the driver for Tasmota. This is how this **free** open source project has grown to such a broad range of capabilities.

Jason2866, 2019-10-13:
Shutter driver support is done

andrethomas2, 2019-10-15:
Closing this issue as associated PR(s) has been merged into development.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-15 by andrethomas2

leonbizco, 2022-08-03:
is there a new function to read  lines from  various gpios inputs  to transduce a rotary absolute encoder? eg , a wind direction in a weather station

barbudor, 2022-08-03:
This issue is closed for 3 years
Please start a new Discussion (not an issue) explaining what you want to achieve