arendst/Tasmota #290: slow which time when pressing the button on Sonoff S20

redviper2100, 2017-03-26:
Hi,

I have a little issue regarding the Sonoff S20 when pushing the button on the plug. It take around one second for the relay to react after pushing the button. If I switch it by MQTT or the built-in http command the relay switches almost instantly.

How can I solve this?

Thanks!

davidelang, 2017-03-26:
look at the logs to see exactly what's happening when you hit the switch, set 
the log level to more debugging

then put the logs here.

On Sun, 26 Mar 2017, redviper2100 wrote:

> Date: Sun, 26 Mar 2017 08:23:44 -0700
> From: redviper2100 <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24fca55cc1326f2d6ea7b1b5c7bc63c16f41d54529792cf0000000114ef9f0
>     092a169ce0cf0105b@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: [arendst/Sonoff-Tasmota] slow which time when pressing the button on
>     Sonoff S20 (#290)
> 
> Hi,
>
> I have a little issue regarding the Sonoff S20 when pushing the button on the plug. It take around one second for the relay to react after pushing the button. If I switch it by MQTT or the built-in http command the relay switches almost instantly.
>
> How can I solve this?
>
> Thanks!
>
>

redviper2100, 2017-03-26:
Here is the output after pressing the button:

> 16:31:28 APP: Multipress 1
16:31:29 MQTT: domoticz/in = {"idx":14,"nvalue":1,"svalue":""}
16:31:29 MQTT: stat/Switch1/RESULT = {"POWER":"ON"}
16:31:29 MQTT: stat/Switch1/POWER = ON
16:31:29 Config: Saved configuration (1364 bytes) to flash at F8 and count 117
16:31:29 RSLT: Receive topic domoticz/out, data size 236, data {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "0001405E",
   "idx" : 14,
   "name" : "Switch1",
   "nvalue" : 1,
   "stype" : "Switch",
   "switchType" : "On/Off",
   "unit" : 1
}

16:31:29 DMTZ: idx 14, nvalue 1

redviper2100, 2017-03-26:
So is it possible to set a smaller delay? For example to wait only a half of a second to decide? 

If so, where could I change the code?

Thank you.

davidelang, 2017-03-26:
search for multiwindow in sonoff.ini

davidelang, 2017-03-27:
I'm going to go ahead and close this, continue to comment or re-open it if you have more questions.

closed 2017-03-27 by davidelang

bbbenji, 2017-08-31:
I have the same issue an found the following

`uint8_t multiwindow[4] = { 0 };       // Max time between button presses to record press count`

Am I understanding correctly that it is already set to 0 wait time? I also find it much easier to update my Sonoff Touches using the prebuilt images therdore I believe there should be no wait time at all as default.

davidelang, 2017-08-31:
no wait time means that all the multipress functions will not work. That's not 
appropriate for the pre-built images.

bbbenji, 2017-08-31:
I understand. Perhaps an option in the UI would be a better idea. Though it's my fault I didn't read the documentation and was unaware of the multipress feature.

For future reference I found the current solution for this. In the console input `SetOption13 1`. This will enables the single press only feature.