arendst/Tasmota #291: Prefix issue in 4.1.0

tam481, 2017-03-26:
Add command Prefix1 to Prefix3 to assign SUB_PREFIX, PUB_PREFIX and PUB_PREFIX2 respectively (#255)

Hello,
I found a bug in relation to modifying the prefix variables in the latest release.

My project name is "TableLamp"
I have mine preset to "cmnd/home/livingroom". Using MQTT-Spy I found that the Sonoff subscribes and publishes to cmnd/home//TableLamp and tele/home//TableLamp"

arendst, 2017-03-26:
As designed. The max prefix length is only 10 characters due to memory restraints. See #255

tam481, 2017-03-26:
oh no. I have to say I agree with @Hogesyx. This was not an issue. Any chance you will consider going back to the previous configuration?

closed 2017-04-23 by davidelang