arendst/Tasmota #293: Unable to upload mode sonoff dual

Blackir62, 2017-03-26:
Hello,

I uploaded Tasmota on a dual sonoff a first time with success and everything worked correctly but since I restart it, impossible to reconnect on.
So I tried to re-load tasmota but without success, I can no longer pass the ESP8266 in upload mode, would I have missed something?

Thank you.

davidelang, 2017-03-27:
getting the device into flashing mode overrides any software on it, so if you are not getting it in flashing mod e you are missing something in your attempt. remember that on the dual, the button is not gpio0, you have to have gpio0 connected to ground as it powers up to be able to flash it

also double check that you haven't reversed tx/rx pins

Blackir62, 2017-03-28:
Thanks, I check all my conection but nothing.
I have the same result on a sonoff s20, unable to put the sonoff in flashing mode.
Do you have clue ?

davidelang, 2017-03-28:
the only reason why you would not get the device in flashing mode is if it powers up without gpio held to ground.

to actually flash it from there you need to have a connection to the serial port on the device and use the appropriate software to spit the data to it at the right speed. We've seen people have driver problems on windows, be telling the software the wrong port, have pins wrong, etc.

davidelang, 2017-04-02:
one other issue you may be running into is that the config is corrupted, see the troubleshooting page in the wiki for how to override or reset the config

closed 2017-04-02 by davidelang