arendst/Tasmota #294: 'TelePeriod' in MQTT messages showing always '0'

agsochi, 2017-03-26:
I set TelePeriod to 20, but after message is sent it changes to 0
```
22:30:14 CMND: TelePeriod
22:30:14 MQTT: stat/sonoffpow1/RESULT = {"TelePeriod":"0"}
22:30:22 CMND: TelePeriod   20
22:30:22 MQTT: stat/sonoffpow1/RESULT = {"TelePeriod":"20"}
22:30:22 MQTT: tele/sonoffpow1/STATE = {"Time":"2017-03-26T22:30:22", "Uptime":0, "Vcc":3.168, "POWER":"ON", "Wifi":{"AP":1, "SSID":"AG_Network", "RSSI":82}}
22:30:22 MQTT: tele/sonoffpow1/ENERGY = {"Time":"2017-03-26T22:30:22", "Yesterday":0.000, "Today":0.012, "Period":0, "Power":0, "Factor":0.00, "Voltage":234, "Current":0.000}
22:30:38 CMND: TelePeriod   
22:30:38 MQTT: stat/sonoffpow1/RESULT = {"TelePeriod":"0"}
```

arendst, 2017-03-27:
Please provide output of command status 0.

agsochi, 2017-03-27:
Here it is:
```
23:13:52 CMND: status 0
23:13:52 MQTT: stat/sonoffpow1/STATUS = {"Status":{"Module":6, "FriendlyName":"sonoffpow1", "Topic":"sonoffpow1", "ButtonTopic":"0", "Subtopic":"POWER", "Power":1, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
23:13:52 MQTT: stat/sonoffpow1/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":2, "Sleep":0, "BootCount":20, "SaveCount":49}}
23:13:52 MQTT: stat/sonoffpow1/STATUS2 = {"StatusFWR":{"Program":"4.1.0", "BuildDateTime":"2017-03-25T22:46:43", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
23:13:52 MQTT: stat/sonoffpow1/STATUS3 = {"StatusLOG":{"Seriallog":0, "Weblog":2, "Syslog":0, "LogHost":"domus1", "SSId1":"AG_Network", "SSId2":"MartiniMan", "TelePeriod":20}}
23:13:52 MQTT: stat/sonoffpow1/STATUS4 = {"StatusMEM":{"ProgramSize":446, "Free":492, "Heap":24, "SpiffsStart":940, "SpiffsSize":64, "FlashSize":4096, "ProgramFlashSize":1024, "FlashChipMode":2}}
23:13:52 MQTT: stat/sonoffpow1/STATUS5 = {"StatusNET":{"Host":"sonoffpow1", "IP":"192.168.1.41", "Gateway":"192.168.1.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.1", "Mac":"5C:CF:7F:A0:36:D8", "Webserver":2, "WifiConfig":3}}
23:13:52 MQTT: stat/sonoffpow1/STATUS6 = {"StatusMQT":{"Host":"192.168.1.2", "Port":1883, "ClientMask":"sonoffpow1", "Client":"sonoffpow1", "User":"sonoffpow1", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
23:13:52 MQTT: stat/sonoffpow1/STATUS7 = {"StatusTIM":{"UTC":"Mon Mar 27 22:13:52 2017", "Local":"Mon Mar 27 23:13:52 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
23:13:52 MQTT: stat/sonoffpow1/STATUS8 = {"StatusPWR":{"Yesterday":0.015, "Today":0.028, "Power":0, "Factor":0.00, "Voltage":236, "Current":0.000}}
23:13:52 MQTT: stat/sonoffpow1/STATUS9 = {"StatusPTH":{"PowerLow":0, "PowerHigh":0, "VoltageLow":0, "VoltageHigh":0, "CurrentLow":0, "CurrentHigh":0}}
23:13:52 MQTT: stat/sonoffpow1/STATUS10 = {"StatusSNS":{"Time":"2017-03-27T23:13:52"}}
```

davidelang, 2017-03-27:
ok, that's showing teleperiod correctly (as 20 in status3), so it looks like 
there is a bug in reporting it with the teleperiod command

arendst, 2017-03-28:
I can't see it fail in my case:
```
16:42:15 MQTT: stat/pow1/RESULT = {"Units":"ON"}
16:42:22 CMND: teleperiod
16:42:22 MQTT: stat/pow1/RESULT = {"TelePeriod":"300 Sec"}
16:44:54 MQTT: tele/pow1/STATE = {"Time":"2017-03-28T16:44:54", "Uptime":62, "Vcc":3.176, "POWER":"ON", "Wifi":{"AP":2, "SSID":"indebuurt2", "RSSI":100}}
16:44:54 MQTT: tele/pow1/ENERGY = {"Time":"2017-03-28T16:44:54", "Yesterday":0.047, "Today":0.033, "Period":0, "Power":0, "Factor":0.00, "Voltage":213, "Current":0.000}
16:45:59 CMND: teleperiod 20
16:45:59 MQTT: stat/pow1/RESULT = {"TelePeriod":"20 Sec"}
16:46:00 MQTT: tele/pow1/STATE = {"Time":"2017-03-28T16:46:00", "Uptime":62, "Vcc":3.176, "POWER":"ON", "Wifi":{"AP":2, "SSID":"indebuurt2", "RSSI":100}}
16:46:00 MQTT: tele/pow1/ENERGY = {"Time":"2017-03-28T16:46:00", "Yesterday":0.047, "Today":0.033, "Period":0, "Power":0, "Factor":0.00, "Voltage":203, "Current":0.000}
16:46:05 CMND: teleperiod
16:46:05 MQTT: stat/pow1/RESULT = {"TelePeriod":"20 Sec"}
16:46:10 CMND: units off
16:46:10 MQTT: stat/pow1/RESULT = {"Units":"OFF"}
16:46:19 CMND: teleperiod 300
16:46:19 MQTT: stat/pow1/RESULT = {"TelePeriod":"300"}
```
I do see that your file size is a lot smaller (446) than mine (467) so you have disabled some features. You also seem to miss Status11 for some reason unclear to me.

Please retry with the pre-compiled version from the release page and report back.

agsochi, 2017-04-01:
Upgraded to 4.1.1 - its ok now.
```
08:27:38 CMND: TelePeriod
08:27:38 MQTT: stat/sonoffpow1/RESULT = {"TelePeriod":"20"}
08:27:55 MQTT: tele/sonoffpow1/STATE = {"Time":"2017-04-01T08:27:55", "Uptime":0, "Vcc":3.170, "POWER":"ON", "Wifi":{"AP":1, "SSID":"AG_Network", "RSSI":76}}
08:27:55 MQTT: tele/sonoffpow1/ENERGY = {"Time":"2017-04-01T08:27:55", "Yesterday":0.031, "Today":0.011, "Period":0, "Power":0, "Factor":0.00, "Voltage":227, "Current":0.000}
08:28:15 MQTT: tele/sonoffpow1/STATE = {"Time":"2017-04-01T08:28:15", "Uptime":0, "Vcc":3.170, "POWER":"ON", "Wifi":{"AP":1, "SSID":"AG_Network", "RSSI":78}}
08:28:15 MQTT: tele/sonoffpow1/ENERGY = {"Time":"2017-04-01T08:28:15", "Yesterday":0.031, "Today":0.011, "Period":0, "Power":0, "Factor":0.00, "Voltage":229, "Current":0.000}
08:28:35 MQTT: tele/sonoffpow1/STATE = {"Time":"2017-04-01T08:28:35", "Uptime":0, "Vcc":3.171, "POWER":"ON", "Wifi":{"AP":1, "SSID":"AG_Network", "RSSI":74}}
08:28:35 MQTT: tele/sonoffpow1/ENERGY = {"Time":"2017-04-01T08:28:35", "Yesterday":0.031, "Today":0.011, "Period":0, "Power":0, "Factor":0.00, "Voltage":227, "Current":0.000}
08:28:55 MQTT: tele/sonoffpow1/STATE = {"Time":"2017-04-01T08:28:55", "Uptime":0, "Vcc":3.170, "POWER":"ON", "Wifi":{"AP":1, "SSID":"AG_Network", "RSSI":74}}
08:28:55 MQTT: tele/sonoffpow1/ENERGY = {"Time":"2017-04-01T08:28:55", "Yesterday":0.031, "Today":0.011, "Period":0, "Power":0, "Factor":0.00, "Voltage":227, "Current":0.000}
08:28:59 CMND: TelePeriod
08:28:59 MQTT: stat/sonoffpow1/RESULT = {"TelePeriod":"20"}
08:29:15 MQTT: tele/sonoffpow1/STATE = {"Time":"2017-04-01T08:29:15", "Uptime":0, "Vcc":3.170, "POWER":"ON", "Wifi":{"AP":1, "SSID":"AG_Network", "RSSI":72}}
08:29:15 MQTT: tele/sonoffpow1/ENERGY = {"Time":"2017-04-01T08:29:15", "Yesterday":0.031, "Today":0.011, "Period":0, "Power":0, "Factor":0.00, "Voltage":226, "Current":0.000}
```
But why in MQTT message always`"Period":0,`?

arendst, 2017-04-01:
I haven't changed any code regarding teleperiod so I can't explain why yours is working fine now.

About Period, it's the energy used during the time between two telemetry messages. In your case it is 20 seconds and with no load it will stay at 0 energy usage. See wiki > Tips > Sonoff Pow for more background information. To see anything here you would need a lot of energy usages or just use a more 'sensable' teleperiod of 300 seconds

closed 2017-04-01 by agsochi