arendst/Tasmota #295: Onewire device, MAX31850K thermocouple amplifier

Beef-well, 2017-03-27:
The MAX31850K is a one wire device that responds to the same commands as a DS18b20. The only difference I know of is the family number starts with 3B. I've tried adding a 3B entry to line 66 and line 174 of xsns_ds18x20.ino. In the previous program I used with this device that was all that needed to be done and it worked. Is there another place that the family name needs to be added?

Beef-well, 2017-03-27:
Ha Ha, NM it works. Just had to add the family number in 3 locations and it picks right up.

 Line 165 of xsns_ds18x20.ino under "Presentation" Added
`
case 0x3B:
      strcpy_P(dsbstype, PSTR("MAX31850K"));
      break;
`
Line 64 CRC check
`
(ds18x20_addr[num_sensors][0]==0x3B
`
Line 155
`
case 0x3B:  // MAX31850K
        t = ((data[1] << 8) + data[0]) * 0.0625;
        if(S) t = ds18x20_convertCtoF(t);
        break;
`

Beef-well, 2017-03-27:
Ok so technically it works, but I end up with this on the WebUI.
[image]

davidelang, 2017-03-27:
that would seem to indicate that it's responding to multiple ids on the one-wire bus.

Beef-well, 2017-03-27:
Well I found that it works normally if Line 165 is left as 
`
case 0x3B: strcpy_P(dsbstype, PSTR("DS18B20")); break;
`
I tried to customize the device type by using
`
case 0x3B: strcpy_P(dsbstype, PSTR("MAX31850K")); break;
`
that seems to put into a loop and adds max number of sensors even though there are only 2.

davidelang, 2017-03-27:
it shouldn't use the same device type, should it?

Beef-well, 2017-03-27:
No, but it works. It just sees another DS18B20 with a different family number.

arendst, 2017-03-29:
Implemented in next release. Please test.

Beef-well, 2017-03-29:
Works beautifully. You are sir are a wizard. Thank you so much.

closed 2017-03-29 by Beef-well