arendst/Tasmota #296: Wemo emulation seems to support only 1 device/relay (ver 4.0.7)

stefanbode, 2017-03-27:
Only the first "Friendly name" seems to be published to WEMO. Is hard coded in xdrv_wemohue.ini line 356 ff.
---
  setup_xml.replace("{x1}", sysCfg.friendlyname[0]);
----
For more than one relay this should loop over the friendly names defined and publish all switches.

stefanbode, 2017-03-27:
For sure two many dependencies to replace this by https://bitbucket.org/xoseperez/fauxmoesp.
But maybe it helps to find out how to get more than one device connected.

altelch, 2017-03-27:
Hi,

Wemo Emulation is supposed for one relay devices only. If you need more, 
please use the HUE Emulation.

Greetings,

      Heiko.

stefanbode, 2017-03-27:
Thanks for the reply. The Wemo is just convenient because you need no external HUB running. AFAIK this is required for HUE. I used the emulation from OpenHAB. Works, but again another component to rely on to work. You can close as enhancement request

altelch, 2017-03-27:
Hi,

no, the HUE Emulation emulates the HUB and presents all devices as HUE 
devices behind the HUB. Works with Echo even with brightness controll if 
you connect a LED strip instead of a normal light.

Greetings,

     Heiko.

stefanbode, 2017-03-30:
you can close. the HUE stuff is working excellent. Also with Alexa. No need to use WeMo.

closed 2017-04-02 by arendst

uros76, 2017-04-29:
Hi. Reading this it seems like HUE works with Alexa without HUE bridge. How did you do that, which Alexa skill did you use?

Magic73, 2020-08-31:
The worst thing to use HUE emu, it that every device will be recognized as Light.

then, if you say: "Alexa, turn on/off all the lights" you turn on/off everithing. event the non-light (like a sonoff switch used to control something else).