arendst/Tasmota #298: Sonoff Basic online flashes

DanielHD80, 2017-03-27:
If I turn on or off my Sonoff Basic with Sonoff-Tasmota by HTTP-Request (http://192.168.1.103/cm?cmnd=Power%20On) it flashes 0.1 seconds and I get the following infos:

COMMANDS1 = {"Commands1":"Status, SaveData, SaveSate, Sleep, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname, Module, Modules, GPIO, GPIOs"}
COMMANDS2 = {"Commands2":"Mqtt, Units, Timezone, LedState, LedPower, TelePeriod"}
COMMANDS3 = {"Commands3":"Power, Light, PowerOnState, PulseTime, BlinkTime, BlinkCount, ButtonRestrict, NtpServer, Weblog, Webserver, WebPassword, Emulation"}
COMMANDS4 = {"Commands4":"DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzSwitchIdx, DomoticzSensorIdx, DomoticzUpdateTimer"}

What is the bug? :(

Daniel

davidelang, 2017-03-27:
that output happens when it receives an invalid command, increase logging to 
'more debug' level and it will show you what commands it's getting and 
generating

then post the logs, starting from just before you send the request.

DanielHD80, 2017-03-27:
Thank you so much! I had an error in my MQTT-Config (but deactivated MQTT!). Now I have corrrected this and it runs! I found the error in log. I didn't found the possibility to increase logging!

Daniel

davidelang, 2017-03-27:
in the webserver, go to configuration and logging
 
you can also do this via serial, mqtt, or the web console with the logging commands listed at https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

or you can set the logging level in user_config*

closed 2017-03-27 by davidelang