arendst/Tasmota #299: openHAB2 + Sonoff switch problem

SzilagyiR, 2017-03-27:
Hi!

I install Mosquitto MQTT Server, openHAB2 on my Raspberry PI3. I use Homebridge, and it works well, but openHAB I have some issues.

Problems:
- The switch works but not always, sometimes not doing anything (MQTT massage not send) after a several switching it works again, and than not again...
- Not showing the Wifi signal strength. (light1_wifi)
- Not showing the Online or Offline state. (light1_on)

The Sonoff webserver log shows MQTT commands are working well:
[image]

[I used andrest wiki to my setup.](https://github.com/arendst/Sonoff-Tasmota/wiki/openHAB )

mqtt.cfg
```
raspberrypi.url=tcp://localhost:1883
raspberrypi.clientId=DVES_%06X
raspberrypi.user=DVES_USER
raspberrypi.pwd=DVES_PASS
```
sonoff.items
```
Switch light1 "Nappali" <light> (LR,gLight)
    { mqtt=">[broker:cmnd/sonoff_1/power:command:*:default],
            <[broker:stat/sonoff_1/POWER:state:default]" }

// A switch being 'ON' as long as the device is reachable
Switch light1_on "Nappali elerheto" (gReachable)
    { mqtt="<[broker:tele/sonoff_1/LWT:state:MAP(reachable.map)]" }

// Wifi Signal Strength in Percent
Number light1_wifi "Nappali wifi [%d %%]" (gRSSI)
    { mqtt="<[broker:tele/sonoff_1/STATE:state:JSONPATH($.Wifi.RSSI)]" }

//A collection of return messages by the Sonoff module
String light1_mqtt "Nappali: [%s]"
    { mqtt="<[broker:tele/sonoff_1/INFO1:state:default],
            <[broker:stat/sonoff_1/STATUS2:state:default],
            <[broker:stat/sonoff_1/RESULT:state:default]" }
```
reachable.map
```
Online=ON
Offline=OFF
```
_default.sitemap
```
sitemap home label="Otthon"
{
Frame label="Lámpák2"
 {
   Switch item=light1
   Text item=light1_on
   Text item=light1_wifi
   Text item=light1_mqtt
 }
```
[image]

davidelang, 2017-03-27:
so is the problem that openhab2 is not sending mqtt commands? or that the sonoff is not sending mqtt commands? have you checked your mqtt server (have it log what it's seeing, or connect to it and subscribe to #)

hamwong, 2017-03-28:
I using as below in OH without issue, just for your information
`Switch 	Sonoff_S20_2 				"Router Power" 		<router>								(GH_Guest_room, gSonoff, gOnOff)	["Switchable"]			{ mqtt=">[mosquitto:cmnd/Sonoff-s20-2/power:command:ON:on],>[mosquitto:cmnd/Sonoff-s20-2/power:command:OFF:off],<[mosquitto:stat/Sonoff-s20-2/POWER:state:default],<[mosquitto:tele/Sonoff-s20-2/STATE:state:JSONPATH($.POWER)]"}
Number 	Sonoff_S20_2_RSSI 			"Router Power RSSI [%d %%]"	<rssi>		(GH_Guest_room, gRSSI)										{ mqtt="<[mosquitto:tele/Sonoff-s20-2/STATE:state:JSONPATH($.Wifi.RSSI)]" }
Number 	Sonoff_S20_2_RSSIAP 		"Router Power RSSI AP [%d %%]"	<wifi>			(GH_Guest_room, gRSSIAP)										{ mqtt="<[mosquitto:tele/Sonoff-s20-2/STATE:state:JSONPATH($.Wifi.AP)]" }
String 	Sonoff_S20_2_Status 		"Router Power [%s]"	<status>							(GH_Guest_room, gUnreach)									{ mqtt="<[mosquitto:tele/Sonoff-s20-2/LWT:state:default"]}`

SzilagyiR, 2017-03-28:
The problem is in the openHab, because everything working fine over Homebridge. It means the MQTT server works fine, and the Sonoff module too.

SzilagyiR, 2017-03-29:
hamwong: I try your code, the result is the same. The switch sometimes work, sometimes not...
RSSI not showing anything.

Here is the code from sonoff.items
```
Switch light2 "Nappali" <light> (LR,gLight)
    { mqtt=">[broker:cmnd/sonoff_1/power:command:ON:on],
            >[broker:cmnd/sonoff_1/power:command:OFF:off],
            <[broker:stat/sonoff_1/POWER:state:default],
            <[broker:tele/sonoff_1/STATE:state:JSONPATH($.POWER)]"}

Number light2_RSSI "Router Power RSSI [%d %%]" <rssi> (GH_Guest_room, gRSSI)
    { mqtt="<[broker:tele/sonoff_1/STATE:state:JSONPATH($.Wifi.RSSI)]" }

Number light2_RSSIAP "Router Power RSSI AP [%d %%]" <wifi> (GH_Guest_room, gRSSIAP)
    { mqtt="<[broker:tele/sonoff_1/STATE:state:JSONPATH($.Wifi.AP)]" }

String light2_Status "Router Power [%s]" <status> (GH_Guest_room, gUnreach)
    { mqtt="<[boroker:tele/sonoff_1/LWT:state:default"]}
```

[image]

SzilagyiR, 2017-03-29:
davidelang: I'm trying to check MQTT server log, I found a mosquitto.log:

```
1490767883: New client connected from 127.0.0.1 as DVES_%06X (c1, k60, u'DVES_USER').
1490767892: New connection from 127.0.0.1 on port 1883.
1490767892: Client DVES_%06X already connected, closing old connection.
1490767892: Client DVES_%06X disconnected.
1490767892: New client connected from 127.0.0.1 as DVES_%06X (c1, k60, u'DVES_USER').
1490767893: New connection from 127.0.0.1 on port 1883.
1490767893: Client DVES_%06X already connected, closing old connection.
1490767893: Client DVES_%06X disconnected.
1490767893: New client connected from 127.0.0.1 as DVES_%06X (c1, k60, u'DVES_USER').
1490767902: New connection from 127.0.0.1 on port 1883.
1490767902: Client DVES_%06X already connected, closing old connection.
1490767902: Client DVES_%06X disconnected.
1490767902: New client connected from 127.0.0.1 as DVES_%06X (c1, k60, u'DVES_USER').
1490767903: New connection from 127.0.0.1 on port 1883.
1490767903: Client DVES_%06X already connected, closing old connection.
1490767903: Client DVES_%06X disconnected.
1490767903: New client connected from 127.0.0.1 as DVES_%06X (c1, k60, u'DVES_USER').
1490767912: New connection from 127.0.0.1 on port 1883.
```

How can I subscribe my MQTT server?

Sorry for fool questions, I'm newbie...

hamwong, 2017-03-29:
It looks like you have double connection on mqtt(multi instance), how do you install you openHAB?

SzilagyiR, 2017-03-29:
[I follow a tutorial from 3. point.](http://raspberrypi4u.blogspot.hu/2016/11/raspberry-pi-smarthome-openhab2.html)

```
echo 'deb http://dl.bintray.com/openhab/apt-repo2 testing main' | sudo tee /etc/apt/sources.list.d/openhab2.list
wget -qO - 'https://bintray.com/user/downloadSubjectPublicKey?username=openhab' | sudo apt-key add –
sudo apt-get update
sudo apt-get install openhab2-offline
sudo systemctl start openhab2.service
sudo systemctl status openhab2.service
sudo systemctl daemon-reload
sudo systemctl enable openhab2.service
```

davidelang, 2017-03-29:
> How can I subscribe my MQTT server?

I use the mqtt dashbard app on my android phone, tell it to connect to the IP of 
the mqtt server (while I'm on wifi) and subscribe to #

SzilagyiR, 2017-03-29:
Thanks, I tried MQTTlens Chrome extension, it works and I see the same that I wrote before.
I'm hiting the switch and sometimes it send MQTT command sometimes not.

davidelang, 2017-03-29:
On Tue, 28 Mar 2017, SzilagyiR wrote:

> Thanks, I tried MQTTlens Chrome extension, it works and I see the same that I wrote before.
> I'm hiting the switch and sometimes it send MQTT command sometimes not.

Ok, just to make sure, you are not hitting it multiple times within a second, 
correct? (this would trigger the multipress features)

Assuming that you are not, we will need to see the detailed logs as it sounds as 
if you are having connectivity problems

set the logging level to 'more debugging' and see if you can catch logs during a 
time that it works and a time that it doesn't. logging via syslog won't work if 
there are network connectivity issues, so you will have to cut-n-paste from the 
web logs or gather the logs via the serial port.

SzilagyiR, 2017-03-30:
Sometimes I hit it, but mostly not :)
Over Homebride I tried to hit as fast as I can, and it works fine.

Where can I fined loging level settings?

davidelang, 2017-03-30:
see the commands page on the wiki to see how to set the logging level via mqtt 
or serial or web console, or you can use the configuration -> logging page of 
the webserver

SzilagyiR, 2017-03-30:
Thanks, I will try it!

I already show the web server log, but the massages from OpenHab not arriveing, only simetimes. If I use Homebridge the massages arrived at all time.

If I watching MQTT massages, is the same. Sometimes OpenHab send MQTT when I push the switch sometimes not.

Tomorrow I will record it to a video to show you.

davidelang, 2017-03-30:
openhab should not be sending anything when you push the button on the sonoff, 
it would be receiving something.

SzilagyiR, 2017-03-30:
I don't pushing the sonoff button, I pushing the virtual switch on the openhab!

davidelang, 2017-03-30:
ahh, ok. If you are not getting messages sent to the mqtt server by openhab when 
you push the switch, you will need to contact the openhab developers. We aren't 
going to be able to troubleshoot that much.

SzilagyiR, 2017-03-31:
OK, I try.
I will write here if I have the solution!

Thanks!

hamwong, 2017-04-01:
have you installed JSONPATH and Transform addon in Openhab?
you need to install it to make it work

SzilagyiR, 2017-04-01:
No, I have not!

No, I use RASPBIAN JESSIE WITH PIXEL.

hamwong, 2017-04-01:
if you pi is intent to only use for openhab, I'll suggest you use openhabian for your pi image, it's almost fully automatic install and so far I don't have any issue which may come from manually installed individual snapshot.

but before you decide go for openhabian, you first should try goto **paperUI> Add-on> Transormations** , install  **JSONPath Transformation and Map Transformation**

hamwong, 2017-04-01:
Regarding WIFI reachable, can you share your reachable.map

SzilagyiR, 2017-04-01:
I installed both of them.

The wifi signat strength working now fine!
The reachable till not working.

My reachable.map
```
Online=ON
Offline=OFF
```

SzilagyiR, 2017-04-01:
Everything works!

[image]

Thanks for your help!!

closed 2017-04-01 by SzilagyiR

elyomr, 2018-05-25:
Hi SzilagyiR 

Could you please tell me how you made Alive item work? I have the same problem you had, it doesn't show anything on the panel, just the name of the element.

Regards