arendst/Tasmota #301: user_config.h not updating

omarusman, 2017-03-28:
Hi,

Just wondering why I'm not able to flash a new firmware to my Sonoff S20?

Scenario:

1. Downloaded the newest release.
2. Upload Tasmota to my device via Arduino (v1.8.2). **It's success**
3. But I found out that my user_config WiFi credentials are wrong.
4. So I opened the downloaded Tasmota files, and updated user_config.h with the correct WiFi credentials.
5. Upload it again.
6. But as I checked the serial monitor, I'm still running the old one. Hence I'm not connected to my WiFi network.

Did I missed something?  I appreciate the help. 

Thanks.

davidelang, 2017-03-28:
please see the troubleshooting page in the wiki

Then let me know if that doesn't answer your question.

gthb96, 2017-03-28:
You may have to reset the Sonoff device to factory settings by holding the switch for 4 seconds down.

omarusman, 2017-03-29:
> _change the value of CFG_HOLDER in user_config and re-flash the device_

This works for me! 👍

closed 2017-04-02 by davidelang