arendst/Tasmota #302: Lost Uptime

KPK75, 2017-03-29:
Has anyone noticed "uptime" counter sometimes not updating ? I've simultaneously restarted all devices and from time to time different one is behind. FW version 4.0.7 but earlier versions had the same issue.

`Mar 28 18:02:30 sonoff5-3250 ESP-MQTT: tele/sonoff5/UPTIME = {"Time":"2017-03-28T18:02:30", "Uptime":19}`
`Mar 28 18:37:35 sonoff5-3250 ESP-MQTT: tele/sonoff5/STATE = {"Time":"2017-03-28T18:37:34", "Uptime":19, "Vcc":2.673, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"SW", "RSSI":24}}`
`Mar 28 19:02:31 sonoff5-3250 ESP-MQTT: tele/sonoff5/UPTIME = {"Time":"2017-03-28T19:02:30", "Uptime":20}`
`Mar 28 19:38:25 sonoff5-3250 ESP-MQTT: tele/sonoff5/STATE = {"Time":"2017-03-28T19:38:25", "Uptime":20, "Vcc":2.670, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"SW", "RSSI":22}}`
`Mar 28 20:38:52 sonoff5-3250 ESP-MQTT: tele/sonoff5/STATE = {"Time":"2017-03-28T20:38:52", "Uptime":20, "Vcc":2.886, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"SW", "RSSI":24}}`
`Mar 28 21:39:15 sonoff5-3250 ESP-MQTT: tele/sonoff5/STATE = {"Time":"2017-03-28T21:39:15", "Uptime":20, "Vcc":2.673, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"SW", "RSSI":32}}`
`Mar 28 22:02:30 sonoff5-3250 ESP-MQTT: tele/sonoff5/UPTIME = {"Time":"2017-03-28T22:02:30", "Uptime":21}`
`Mar 28 22:39:46 sonoff5-3250 ESP-MQTT: tele/sonoff5/STATE = {"Time":"2017-03-28T22:39:45", "Uptime":21, "Vcc":2.889, "POWER1":"OFF", "POWER2":"OFF", "Wifi":{"AP":1, "SSID":"SW", "RSSI":32}}`

arendst, 2017-03-29:
Fixed in next release

KPK75, 2017-03-29:
Great. Thanks :)

closed 2017-03-29 by KPK75