arendst/Tasmota #305: Extra switchs and domoticz

gajotnt, 2017-03-29:
Is it possible to when enabling a GPIO to function has a switch to add in the domoticz page the extra switch?
The sonoff basic has 4 extra GPIO, why not enable an option to when using those extra GPIO have extra IDX settings in domoticz setup page?

arendst, 2017-04-01:
Implemented in next release

gajotnt, 2017-04-03:
I Love you! :D

gajotnt, 2017-04-04:
Got this partialy working :) 
here is the console log:

```
14:22:50 MQTT: cmnd/sonoff/POWER1 = ON
14:22:50 MQTT: domoticz/in = {"idx":6,"nvalue":1,"svalue":""}
14:22:50 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
14:22:50 MQTT: stat/sonoff/POWER = ON
14:22:52 MQTT: cmnd/sonoff/POWER1 = OFF
14:22:52 MQTT: domoticz/in = {"idx":6,"nvalue":0,"svalue":""}
14:22:52 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
14:22:52 MQTT: stat/sonoff/POWER = OFF
14:22:57 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"On"}
14:23:00 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"On"}
14:23:06 CMND: SwitchMode
14:23:06 MQTT: stat/sonoff/RESULT = {"SwitchMode1":0}
14:23:09 CMND: SwitchTopic
14:23:09 MQTT: stat/sonoff/RESULT = {"SwitchTopic":"sonoff"}
```
So i have a switch controling the relay (idx:6) and another switch controling idx:7, but it only turns it on, doesnt send the command to turn off, when it should turn it off, it sends the command to turn on again, is this normal behaviour? Maybe im doing something wrong
Switchmode 0
SwitchTopic 1

arendst, 2017-04-04:
Using switchtopic 1 is special as it sets the switchtopic the same as your sonoff's topic being in your case sonoff. Set switchtopic to anything different than 0 or 1 and it should work as expected.

gajotnt, 2017-04-04:
Confirm working :D 
was having some double toggles when using one of the serial gpio but remenbered i had to turn off serial logs.
Working great now :D 
Were can i make a donation? :D

gajotnt, 2017-04-04:
Arrrrghhhh when i have a wire connected to the TX or RX pin i keep getting switch triggers without doing anything, tought it was from the serial log but it isnt.

```
15:48:55 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:48:55 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:48:56 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:48:56 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:49:06 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:49:10 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:49:11 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:49:15 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:49:21 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:49:21 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:49:23 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:49:23 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:49:24 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:49:24 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:49:29 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
15:49:29 MQTT: domoticz/in = {"command":"switchlight","idx":7,"switchcmd":"Toggle"}
```
Do i have to solder somekind of resistor somewere? :/

arendst, 2017-04-04:
You'll have to add pull up resistors as these gpios are set to serial communication. See other issues here how other people handled this.

closed 2017-04-04 by gajotnt