arendst/Tasmota #306: External button

gemitro, 2017-03-29:
Does anyone know to which pins i must solder a cable in order to  toggle the relay of Sonoff TH16 using an external button?

davidelang, 2017-03-29:
you can use any available gpio and define it as switch1

gemitro, 2017-03-29:
Should i connect the other cable to ground.

davidelang, 2017-03-29:
yes, wire the switch from the gpio pin to ground

closed 2017-04-02 by davidelang

spacefolder, 2017-11-26:
Hi Guys! Sorry for reopening the thread, but it's a very related question. I have access to GPIO 04 on a TH16 (soldered to the available pad). I can use an external switch without problems, but I can not use it a a push button. (Button 2,3 and 4 don't seem to work) How can I achieve this? Thanks, and cheers!

davidelang, 2017-11-26:
when you say it doesn't work, what is happening?

what do you expect to happen instead?