arendst/Tasmota #307: asking for password

jesusvallejo, 2017-03-29:
one of my sonoff started asking for a user and a pass eventhhough i did not set up it, is there any way to reset it without flashing( its hard to do it because of location)?? it also does not respond to mqtt commands.

thank you

davidelang, 2017-03-29:
hold down the button for 4 seconds and it will revert to the compiled-in config

jesusvallejo, 2017-03-29:
Tryed, nothing changed.any other idea??

davidelang, 2017-03-29:
if resetting the config doesn't solve it, then someone may have flashed the 
device 'for you'.

I would not trust anything on it and would re-flash it, and then lock down the 
device with a password, you may have someone playing games on your network.

davidelang, 2017-04-23:
closing this issue as there has been no response, please reopen if this is still a problem

closed 2017-04-23 by davidelang

mittelfeldmotor, 2017-07-26:
I have the same problem. The sonoff worked for about 8 weeks. 2 Days ago i got the problem that the sonoff didn´t work. So I tried to open the webpage to check the config. I am asked for an user and a pw. I didn´t set any. So I tried to flash it new. The flashing went fine no problems, but the config didn´t change. can someone help me?

zizebra, 2017-07-26:
you probably need to erase your flash before loading new firmware

arendst, 2017-07-26:
As answered in another issue before, simply change the password using mqtt...

czyz, 2017-11-20:
Hah, I've run into this same issue. i think there's some sort of memory corruption going on. My Sonoff Basic running Tasmota was working just fine immediately after configuraiton, but after unplugging it and then plugging it back into power in a different room it was nonresponsive. I held down the button to reset the configuration, and the SSID "ITEAD-######c##c" appeared. I connected to this SSID and a captive portal showed up, but it looked pretty garbled (screenshot attached). 
[image]

I managed to set it to use the SSID name and credentials of my local wifi, but it never shows up on the network and is no longer accessible via MQTT, and the ITEAD* SSID remains available. If I connect to this ITEAD* SSID I do get a captive portal that shows my new settings (though its title still displays garbled characters). If I hold down the button to restore to the original config, I just get back to the garbled portal seen in the attached screenshot.

One of the blank and unnamed buttons on the garbled screenshot sometimes pops up an authentication prompt that sounds like the problem described at the start of this thread. It may have an unrelated cause, given that the initial poster doesn't mention any garbled characters.
[image]

federic0, 2018-04-09:
@czyz, did you get to solve the issue you describe above? I am having the exact same issue and cannot get my SonOff back to a functional state.

csshull, 2020-12-20:
I am also having the problem but is a little different.  If I try to use a browser to get to the tasmota flashed device, I am asked for a user name and password. The difference is the device still works.  When I press a button, it shows up in the MQTT and I can toggle the output plug.  Works ok but I just can't connect to it to reconfig.  Whe I hole down the button, the device appears to reset but I cannot re-flash and if replugged, it goes back to working.