arendst/Tasmota #308: Multiple DHT22 sensors on webpage possible?

HellGL, 2017-03-29:
Hi, 
I have a TH10 module with two DHT22 sensors connected to GPIO14 and GPIO4. They both work fine on their GPIO but only one of them (GPIO14) show up on the main page. There is also only one log entry on the console. Is there a way to see both sensors on the main web page or at least in the console log?

Best,
HellG

davidelang, 2017-03-29:
no, currently tasmota only allows a single instance of a particular sensor (with 
the one exception that multiple one-wire sensors can be on a single gpio)

HellGL, 2017-03-29:
Is this a limitation to be overcome in the future or what could I do?

davidelang, 2017-03-30:
changing this would be a fairly significant architecture change to the code, so 
while I think it's a good idea, I also don't think it's going to be fixed any 
time soon.

there is some documentation on running multiple sensors on one gpio, I don't 
know if that applies to these sensors, and I don't know where it's documented. 
Sorry I can't be more help. :-(

On Wed, 29 Mar 2017, HellGL wrote:

> Is this a limitation to be overcome in the future or what could I do?

closed 2017-04-02 by davidelang

HellGL, 2017-04-06:
Hi,
I had a closer look into the code but I needed to convert it first into a Arduino Project for Eclipse because I needed some help from a IDE with more capabilities than the Arduino IDE. Although my project compiles without errors (after some changes, but that is another story) I always get an "File upload buffer miscompare" when I try to upload the bin file (but that is also another story...). However I was able to add some code to access up to 3 DHT sensors attached to the GPIO pins of my Sonoff (TH10). 
I attached a patch to the current git with my changes.
[multi_dht2.patch.gz](https://github.com/arendst/Sonoff-Tasmota/files/902143/multi_dht2.patch.gz)
To better distinguish the sensors I extended the naming (DHT_22, DHT_11 etc) with the number of the GPIO pin (=> DHT22_14). Please have a look at the patch. Comments are welcome.